devsim/devsim #4: Question about DevSim Compile CGNS:

chihiro0019, 2015-05-13:
Sorry to bother you, but I am trying to compile DevSim and met a problem.
How can I define TCL TK paths ?

7. cmake -DBUILD_CGNSTOOLS=ON -DCMAKE_INSTALL_PREFIX=$PWD/../../cgnslib ..

-- Could NOT find TCL  (missing:  TCL_INCLUDE_PATH)
-- Could NOT find TCLTK  (missing:  TCL_INCLUDE_PATH TK_INCLUDE_PATH)
-- Could NOT find TK  (missing:  TK_INCLUDE_PATH)
CMake Error at src/cgnstools/CMakeLists.txt:48 (message):
  The path and library needs to be defined for:

      TCL;TK

-- Configuring incomplete, errors occurred!

juanesanchez, 2015-05-13:
>    1. cmake -DBUILD_CGNSTOOLS=ON

the easiest thing may be to set BUILD_CGNSTOOLS=OFF

Juan

chihiro0019, 2015-05-13:
Thank you for your advice. I could make CGNS. Then,
I tried to go to

Compile SYMDIFF

1. cd external/symdiff
2. sh scripts/setup_ubuntu_12.04.sh
3. cd linux_x86_64_release
4. make -j2
5. ctest -j2

But, there is no file 

$ pwd
/home/s037156b/MyC/DeviceSim/DevSim/devsim-master/external
$ ls
.            cgnslib_3.1.4         superlu_osx_10.10.sh   superlu_ubuntu_12.04.sh           symdiff
..           cgnslib_3.1.4.tar.gz  superlu_osx_10.9.sh    superlu_ubuntu_12.04_gcc-4.7.sh
SuperLU_4.3  superlu_4.3.tar.gz    superlu_redhat_6.5.sh  superlu_ubuntu_12.04_icc_2013.sh

$ pwd
/home/s037156b/MyC/DeviceSim/DevSim/devsim-master/external/symdiff
$ ls
.  ..
Is anything wrong ?

juanesanchez, 2015-05-13:
Hi,

Please do:
git submodule init
git submodule update

in the devsim directory.

Regards,

Juan

chihiro0019, 2015-05-13:
Thank you for your reply.
But I'm not familiar with "git". I found some web pages about it.
However could you tell me the right way to install "git" for DevSim ?
The command "git" is not in our machines.

chihiro0019, 2015-05-13:
I tried this instruction below according to "git" web site, however
I am using computer farm and I can not be the root.
Maybe I should ask our system manager to do it ?

$ yum install curl-devel expat-devel gettext-devel  openssl-devel zlib-devel

You need to be root to perform this commnd.

juanesanchez, 2015-05-13:
git is difficult to compile.  Download the symdiff package from:
http://github.com/devsim/symdiff

and put the symdiff directory in the external directory of devsim.

Juan

chihiro0019, 2015-05-14:
Thank you for your reply.

I donwloaded http://github.com/devsim/symdiff and put it as below.

$ pwd
/home/s037156b/MyC/DeviceSim/DevSim/devsim-master/external
$ ls
.            cgnslib_3.1.4         superlu_osx_10.10.sh   superlu_ubuntu_12.04.sh           symdiff
..           cgnslib_3.1.4.tar.gz  superlu_osx_10.9.sh    superlu_ubuntu_12.04_gcc-4.7.sh   symdiff-master
SuperLU_4.3  superlu_4.3.tar.gz    superlu_redhat_6.5.sh  superlu_ubuntu_12.04_icc_2013.sh  symdiff-master.zip

In symdiff directory, there is no file.
I tried to install symdiff from symdiff-master directory, but

[s037156b@xhz352 symdiff-master]$ cmake g++ flex bison python tcl tcl-dev python-dev
CMake Error: The source directory "/home/s037156b/MyC/DeviceSim/DevSim/devsim-master/external/symdiff-master/python-dev" does not exist.

What should I do ?

chihiro0019, 2015-05-14:
According to our system manager, we have a git
 /tooltop/vendor/utility/git_1.8.5.4

I tried to install DEVSIM

$ git clone git://github.com/devsim/devsim.git
Cloning into 'devsim'...
fatal: Unable to look up github.com (port 9418) (Name or service not known)

This may be a different problem about networking ?

juanesanchez, 2015-05-14:
You may replace the external/symdiff directory with the one you downloaded.
rm -rf symdiff
mv symdiff-master symdiff

then follow the instructions in symdiff/INSTALL.  You may need to use:
symdiff/scripts/setup_redhat_6.5.sh 
<https://github.com/devsim/symdiff/blob/master/scripts/setup_redhat_6.5.sh>

so that it is appropriate for redhat 5.  You may have problems with the 
c++ compiler, because you need to support advanced c++ features not 
available in redhat 5.

Regards,

Juan

juanesanchez, 2015-05-14:
This sounds like a networking issue.

chihiro0019, 2015-05-14:
I set up proxy and the situation was changed. But still doesn't work.

$ git config --global http.proxy proxy.xxx.xxx.co.jp:10080
$ git submodule init
fatal: Not a git repository (or any parent up to mount point /home/s037156b)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).

juanesanchez, 2015-05-14:
You would need to clone devsim first so that you have a .git directory 
as part of the software.

1. git clone git://github.com/devsim/devsim.git
2. cd devsim
3. git submodule init
4. git submodule update

chihiro0019, 2015-05-14:
Yes, thank you, I tried it first. But it didn't work.

$ git config --global http.proxy proxy.xxxx.xxxx.co.jp:10080
$ git clone git://github.com/devsim/devsim.git
Cloning into 'devsim'...
fatal: Unable to look up github.com (port 9418) (Name or service not known)

juanesanchez, 2015-05-14:
You can also use github over a firewall using http:
https://github.com/blog/92-http-cloning

So try:
git clone http://github.com/devsim/devsim.git
or
git clone https://github.com/devsim/devsim.git

chihiro0019, 2015-05-15:
Oh, thank you. I went forward a little, and I still have another problem shown below.
The last one 'git submodule update' seems to fail.

$ git clone http://github.com/devsim/devsim.git
Cloning into 'devsim'...
remote: Counting objects: 1767, done.
remote: Total 1767 (delta 0), reused 0 (delta 0), pack-reused 1767
Receiving objects: 100% (1767/1767), 30.69 MiB | 263.00 KiB/s, done.
Resolving deltas: 100% (1067/1067), done.
Checking connectivity... done.
Checking out files: 100% (785/785), done.

$ git submodule init
Submodule 'external/symdiff' (git://github.com/devsim/symdiff.git) registered for path 'external/symdiff'

$ git submodule update
Cloning into 'external/symdiff'...
fatal: Unable to look up github.com (port 9418) (Name or service not known)
Clone of 'git://github.com/devsim/symdiff.git' into submodule path 'external/symdiff' failed

juanesanchez, 2015-05-15:
edit the following file:
devsim/.git/config

and change the line for symdiff to use http:// instead of git://

chihiro0019, 2015-05-15:
Thank you, I tried 2 cases,  // instead of git://  and  //instead of git:// 
But both didn't work.

$ git submodule update
Cloning into 'external/symdiff'...
fatal: unable to access 'http:// instead of git://': The requested URL returned error: 406
Clone of 'http:// instead of git://' into submodule path 'external/symdiff' failed
$ git submodule update
Cloning into 'external/symdiff'...
fatal: unable to access 'http://instead of git://': The requested URL returned error: 406
Clone of 'http://instead of git://' into submodule path 'external/symdiff' failed

juanesanchez, 2015-05-15:
try:
cd devsim/external
rm -rf symdiff
git clone

chihiro0019, 2015-05-18:
Thank you for your kind support.
I still have some problems, but I give up to continue.
I heard that the other guys in my company have already installed this.
They are using local machine on which they can be the root unlike my case.
It might be the largest problem for me. I apologize it and thank you for your
very kind support, I appreciate you.

closed 2015-05-18 by chihiro0019