devsim/devsim #7: Error during compiling CGNS

chihiro0019, 2015-08-11:
I met compile errors, please advice.
I got OP log shown at the bottom, but, TK has been installed.

------------------------------------------------------------------------- TK installation
[root@JPC20320148 devsim-master]# yum install tk
読み込んだプラグイン:fastestmirror, product-id, refresh-packagekit, rhnplugin, security, subscription-manager
This system is receiving updates from RHN Classic or RHN Satellite.
インストール処理の設定をしています
Loading mirror speeds from cached hostfile
 * base: ftp.hosteurope.de
 * epel: ftp.cuhk.edu.hk
 * extras: ftp.hosteurope.de
 * rpmforge: apt.sw.be
 * updates: ftp.hosteurope.de
パッケージ 1:tk-8.5.7-5.el6.x86_64 はインストール済みか最新バージョンです
何もしません
---------------------------------------------------------------------------------- TK installation already done

And I could not find the file "CMake Error at src/cgnstools/CMakeLists.txt:52 (message)"

-------------------------------------------------------------------- ls src/cgnstools/
[root@JPC20320148 cgnstools]# pwd
/install/DevSim/devsim-master/external/cgnslib_3.1.4/build/src/cgnstools
[root@JPC20320148 cgnstools]# ls
.  ..  CMakeFiles  cgconfig
[root@JPC20320148 cgnstools]# ls CMakeFiles/
.  ..
[root@JPC20320148 cgnstools]# 
-------------------------------------------------------------------------  ls src/cgnstools/

--------------------------------------------------------------------------------------------------------- OP log
[root@JPC20320148 external]# ls
CMakeCache.txt  cgnslib_3.1.4-2.tar.gz  superlu_osx_10.9.sh      superlu_ubuntu_12.04_gcc-4.7.sh   testing
CMakeFiles      src                     superlu_redhat_6.5.sh    superlu_ubuntu_12.04_icc_2013.sh
cgnslib_3.1.4   superlu_osx_10.10.sh    superlu_ubuntu_12.04.sh  symdiff
[root@JPC20320148 external]# cd cgnslib_3.1.4
[root@JPC20320148 cgnslib_3.1.4]# ls
CMakeLists.txt  cmake_uninstall.cmake.in  fortran_test  install.txt  pcgns       readme.txt
changelog       doc                       install.lyx   license.txt  readme.lyx  src
[root@JPC20320148 cgnslib_3.1.4]# mkdir build
[root@JPC20320148 cgnslib_3.1.4]# cd build
[root@JPC20320148 build]# cmake -DBUILD_CGNSTOOLS=ON -DCMAKE_INSTALL_PREFIX=$PWD/../../cgnslib ..
-- The C compiler identification is GNU 4.4.7
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Looking for sys/types.h
-- Looking for sys/types.h - found
-- Looking for stdint.h
-- Looking for stdint.h - found
-- Looking for stddef.h
-- Looking for stddef.h - found
-- Check size of long
-- Check size of long - done
-- Check size of off_t
-- Check size of off_t - done
-- Looking for XOpenDisplay in /usr/lib64/libX11.so;/usr/lib64/libXext.so
-- Looking for XOpenDisplay in /usr/lib64/libX11.so;/usr/lib64/libXext.so - found
-- Looking for gethostbyname
-- Looking for gethostbyname - found
-- Looking for connect
-- Looking for connect - found
-- Looking for remove
-- Looking for remove - found
-- Looking for shmat
-- Looking for shmat - found
-- Found X11: /usr/lib64/libX11.so
-- Found Tclsh: /usr/bin/tclsh (found version "8.5") 
-- Could NOT find TCL (missing:  TCL_INCLUDE_PATH) 
-- Could NOT find TCLTK (missing:  TCL_INCLUDE_PATH TK_INCLUDE_PATH) 
-- Could NOT find TK (missing:  TK_INCLUDE_PATH) 
-- Found OpenGL: /usr/lib64/libGL.so  
CMake Error at src/cgnstools/CMakeLists.txt:52 (message):
  The path and library needs to be defined for:

      TCL;TK

-- Configuring incomplete, errors occurred!
See also "/install/DevSim/devsim-master/external/cgnslib_3.1.4/build/CMakeFiles/CMakeOutput.log".
[root@JPC20320148 build]# pwd
/install/DevSim/devsim-master/external/cgnslib_3.1.4/build
[root@JPC20320148 build]# pwd
/install/DevSim/devsim-master/external/cgnslib_3.1.4/build
[root@JPC20320148 build]# ls
CMakeCache.txt  CMakeFiles  CPackConfig.cmake  CPackSourceConfig.cmake  cmake_uninstall.cmake  src
[root@JPC20320148 build]# 
------------------------------------------------------------------------------------------------------------- OP log

chihiro0019, 2015-08-11:
Sorry, the file /install/DevSim/devsim-master/external/cgnslib_3.1.4/src/cgnstools/CMakeLists.txt  is there. But, I can't know what is wrong.

CMake Error at src/cgnstools/CMakeLists.txt:52 (message):
The path and library needs to be defined for:

  TCL;TK

-- Configuring incomplete, errors occurred!

juanesanchez, 2015-08-11:
Please install:
tk-devel
tcl-devel

chihiro0019, 2015-08-11:
Thank you, I got a progress, but then I met another problem shown below.

[root@JPC20320148 build]# cmake -DBUILD_CGNSTOOLS=ON -DCMAKE_INSTALL_PREFIX=$PWD/../../cgnslib ..
-- Found TCL: /usr/lib64/libtcl8.5.so  
-- Found TCLTK: /usr/lib64/libtcl8.5.so  
-- Found TK: /usr/lib64/libtk8.5.so  
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
X11_Xmu_LIB (ADVANCED)
    linked by target "plotwish" in directory /install/DevSim/devsim-master/external/cgnslib_3.1.4/src/cgnstools/cgnsplot

-- Configuring incomplete, errors occurred!
See also "/install/DevSim/devsim-master/external/cgnslib_3.1.4/build/CMakeFiles/CMakeOutput.log".
[root@JPC20320148 build]#

juanesanchez, 2015-08-11:
Please install:
libXmu
libXmu-devel

chihiro0019, 2015-08-11:
Still have some problems.

Installing ibXmu-devel seems failed.

----------------------------------------------------------------------------------------------------- installation ibXmu-devel
[root@JPC20320148 devsim-master]# yum install libXmu ibXmu-devel
読み込んだプラグイン:fastestmirror, product-id, refresh-packagekit, rhnplugin, security, subscription-manager
This system is receiving updates from RHN Classic or RHN Satellite.
インストール処理の設定をしています
Loading mirror speeds from cached hostfile
 * base: centos.mirror.iweb.ca
 * epel: epel.mirror.srv.co.ge
 * extras: centos.mirror.iweb.ca
 * rpmforge: apt.sw.be
 * updates: centos.mirror.iweb.ca
パッケージ libXmu-1.1.1-2.el6.x86_64 はインストール済みか最新バージョンです    libXmu-1.1.1-2.el6.x86_64 is O.K.
パッケージ ibXmu-devel は利用できません。   ibXmu-devel  can not be used
何もしません
-------------------------------------------------------------------------------------------------- installation ibXmu-devel

----------------------------------------------------------------------------------------------- Cmake 
[root@JPC20320148 build]# cmake -DBUILD_CGNSTOOLS=ON -DCMAKE_INSTALL_PREFIX=$PWD/../../cgnslib ..
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
X11_Xmu_LIB (ADVANCED)
    linked by target "plotwish" in directory /install/DevSim/devsim-master/external/cgnslib_3.1.4/src/cgnstools/cgnsplot

-- Configuring incomplete, errors occurred!
See also "/install/DevSim/devsim-master/external/cgnslib_3.1.4/build/CMakeFiles/CMakeOutput.log 
------------------------------------------------------------------------------------- Cmake

juanesanchez, 2015-08-11:
did you install libXmu as well?  If you want skip this part, just use BUILD_CGNSTOOLS=OFF

chihiro0019, 2015-08-11:
Yes, I could install  libXmu  but not  ibXmu-devel   as shown below.
(some parts are in Japanese) 

[root@JPC20320148 build]# yum install libXmu
読み込んだプラグイン:fastestmirror, product-id, refresh-packagekit, rhnplugin, security, subscription-manager
This system is receiving updates from RHN Classic or RHN Satellite.
インストール処理の設定をしています
Loading mirror speeds from cached hostfile
 * base: centos.mirror.iweb.ca
 * epel: epel.mirror.srv.co.ge
 * extras: centos.mirror.iweb.ca
 * rpmforge: apt.sw.be
 * updates: centos.mirror.iweb.ca
パッケージ libXmu-1.1.1-2.el6.x86_64 はインストール済みか最新バージョンです   => already installed
何もしません
[root@JPC20320148 build]# yum install ibXmu-devel
読み込んだプラグイン:fastestmirror, product-id, refresh-packagekit, rhnplugin, security, subscription-manager
This system is receiving updates from RHN Classic or RHN Satellite.
インストール処理の設定をしています
Loading mirror speeds from cached hostfile
 * base: centos.mirror.iweb.ca
 * epel: epel.mirror.srv.co.ge
 * extras: centos.mirror.iweb.ca
 * rpmforge: apt.sw.be
 * updates: centos.mirror.iweb.ca
パッケージ ibXmu-devel は利用できません。  => not available
エラー: 何もしません

chihiro0019, 2015-08-11:
Now, I found a typo,  ibXmu-devel  should be libXmu-devel

And I could cmake !  Thank you so much.

[root@JPC20320148 build]#  cmake -DBUILD_CGNSTOOLS=ON -DCMAKE_INSTALL_PREFIX=$PWD/../../cgnslib ..
-- Looking for IceConnectionNumber in ICE
-- Looking for IceConnectionNumber in ICE - found
-- Configuring done
-- Generating done
-- Build files have been written to: /install/DevSim/devsim-master/external/cgnslib_3.1.4/build

closed 2015-08-11 by chihiro0019