devsim/devsim #8: Compiling SYMDIFF

chihiro0019, 2015-08-12:
I think this is caused by a simple misunderstanding, but could you help me ?

[root@JPC20320148 symdiff]# pwd
/install/DevSim/devsim-master/external/symdiff
[root@JPC20320148 symdiff]# ls ../../scripts/setup_ubuntu_12.04.sh
../../scripts/setup_ubuntu_12.04.sh
[root@JPC20320148 symdiff]# sh ../../scripts/setup_ubuntu_12.04.sh
mkdir:   `linux_x86_64_debug'  already file exists
CMake Error: The source directory "/install/DevSim/devsim-master/external/symdiff" does not appear to contain CMakeLists.txt.
Specify --help for usage, or press the help button on the CMake GUI.
mkdir:  `linux_x86_64_release'  already file exists
CMake Error: The source directory "/install/DevSim/devsim-master/external/symdiff" does not appear to contain CMakeLists.txt.
Specify --help for usage, or press the help button on the CMake GUI.

juanesanchez, 2015-08-12:
in the top level devsim directory, you need to do:
git submodule init
git submodule update

This will populate the symdiff directory properly.

next:
cd external/symdiff
sh scripts/setup_ubuntu_12.04.sh

and follow the remaining instructions in the INSTALL file.  You may be better off using:
setup_redhat_6.5.sh

if you are using a RedHat system.

chihiro0019, 2015-08-12:
Thank you,  I did

> cd /install/DevSim/devsim-master
> git submodule init
> git submodule update

But,

> cd /devsim-master/external/symdiff
> sh scripts/setup_ubuntu_12.04.sh

Do I type the above at /devsim-master/external/symdiff ?
There is no scripts directory under it.

> [root@JPC20320148 symdiff]# pwd
> /install/DevSim/devsim-master/external/symdiff
> [root@JPC20320148 symdiff]# ls
.>   ..  linux_x86_64_debug  linux_x86_64_release

I can find it at

> [root@JPC20320148 scripts]# pwd
> /install/DevSim/devsim-master/scripts
> [root@JPC20320148 scripts]# ls
> .   setup_osx_10.10.sh  setup_redhat_6.5.sh    setup_ubuntu_12.04_gcc-4.7.sh   
> setup_ubuntu_14.04.sh  ..  setup_osx_10.9.sh   setup_ubuntu_12.04.sh  
> setup_ubuntu_12.04_icc_2013.sh setup_windows.sh

Then, I met the same error

> sh /install/DevSim/devsim-master/scripts/setup_ubuntu_12.04.sh
> mkdir:  `linux_x86_64_debug' file exists
> CMake Error: The source directory "/install/DevSim/devsim-master/external/symdiff" does not 
> appear to contain CMakeLists.txt.
> Specify --help for usage, or press the help button on the CMake GUI.
> mkdir:  `linux_x86_64_release'  file exists
> CMake Error: The source directory "/install/DevSim/devsim-master/external/symdiff" does not 
> appear to contain CMakeLists.txt.
> Specify --help for usage, or press the help button on the CMake GUI.

juanesanchez, 2015-08-12:
How did you get DEVSIM?  If you did not use:
git clone

you may need to download symdiff from:
http://github.com/devsim/symdiff

and put it in the external directory manually.

Juan

chihiro0019, 2015-08-12:
Thank you.  This is what I did at first.

[root@JPC20320148 devsim-master]# git config --global http.proxy http://proxy-sen.noc.sony.co.jp:10080
[root@JPC20320148 devsim-master]# git clone https://github.com/devsim/devsim.git
fatal: destination path 'devsim' already exists and is not an empty directory.

But, anyway I will try to download from http://github.com/devsim/symdiff

chihiro0019, 2015-08-12:
Hi, I downloaded http://github.com/devsim/symdiff.

symdiff-master.zip       symdiff-master

Then, copied symdiff-master to symdiff,  then,

# sh scripts/setup_redhat_6.5.sh

It looks like finished, is this O.K. ?

chihiro0019, 2015-08-12:
Dear,  juanesanchez 

I have a new problem in compiling DEVSIM at step 3  make -j2

Compile DEVSIM
cd /install/DevSim/devsim-master/
1. sh scripts/setup_ubuntu_12.04.sh	setup_redhat_6.5.sh
2. cd linux_x86_64_release
3. make -j2

Here is the operation error log.

[root@JPC20320148 devsim-master]# cd linux_x86_64_release
[root@JPC20320148 linux_x86_64_release]# ls
.  ..  CMakeCache.txt  CMakeFiles  CTestTestfile.cmake  Makefile  cmake_install.cmake  src  testing

[root@JPC20320148 linux_x86_64_release]# make -j2
Scanning dependencies of target models
Scanning dependencies of target Geometry
[  1%] Building CXX object src/Geometry/CMakeFiles/Geometry.dir/Device.o
[  2%] Building CXX object src/models/CMakeFiles/models.dir/NodeModel.o
[  2%] Building CXX object src/models/CMakeFiles/models.dir/EdgeModel.o
[  2%] Building CXX object src/Geometry/CMakeFiles/Geometry.dir/Region.o
[  3%] Building CXX object src/models/CMakeFiles/models.dir/InterfaceNodeScalarData.o
[  3%] Building CXX object src/models/CMakeFiles/models.dir/NodeScalarData.o
[  4%] Building CXX object src/models/CMakeFiles/models.dir/EdgeScalarData.o
[  4%] Building CXX object src/models/CMakeFiles/models.dir/TriangleEdgeScalarData.o
[  5%] Building CXX object src/models/CMakeFiles/models.dir/NodeSolution.o
[  5%] Building CXX object src/models/CMakeFiles/models.dir/EdgeSubModel.o
[  6%] Building CXX object src/models/CMakeFiles/models.dir/EdgeFromNodeModel.o
[  6%] Building CXX object src/models/CMakeFiles/models.dir/InterfaceNodeModel.o
[  7%] Building CXX object src/Geometry/CMakeFiles/Geometry.dir/Edge.o
[  8%] Building CXX object src/models/CMakeFiles/models.dir/TriangleEdgeModel.o
[  8%] Building CXX object src/Geometry/CMakeFiles/Geometry.dir/Node.o
[  9%] Building CXX object src/Geometry/CMakeFiles/Geometry.dir/Coordinate.o
[  9%] Building CXX object src/Geometry/CMakeFiles/Geometry.dir/Triangle.o
[  9%] Building CXX object src/models/CMakeFiles/models.dir/TriangleEdgeSubModel.o
[ 10%] Building CXX object src/Geometry/CMakeFiles/Geometry.dir/Contact.o
[ 11%] Building CXX object src/models/CMakeFiles/models.dir/TriangleEdgeFromEdgeModel.o
[ 11%] Building CXX object src/models/CMakeFiles/models.dir/TriangleEdgeFromEdgeModelDerivative.o
[ 11%] Building CXX object src/Geometry/CMakeFiles/Geometry.dir/Permutation.o
[ 12%] Building CXX object src/Geometry/CMakeFiles/Geometry.dir/Interface.o
[ 13%] Building CXX object src/models/CMakeFiles/models.dir/TetrahedronEdgeScalarData.o
[ 13%] Building CXX object src/models/CMakeFiles/models.dir/TetrahedronEdgeModel.o
[ 13%] Building CXX object src/Geometry/CMakeFiles/Geometry.dir/GradientField.o
[ 14%] Building CXX object src/models/CMakeFiles/models.dir/TetrahedronEdgeSubModel.o
[ 15%] Building CXX object src/Geometry/CMakeFiles/Geometry.dir/TriangleElementField.o
[ 15%] Building CXX object src/models/CMakeFiles/models.dir/TetrahedronEdgeFromEdgeModel.o
[ 16%] Building CXX object src/models/CMakeFiles/models.dir/TetrahedronEdgeFromEdgeModelDerivative.o
[ 16%] Building CXX object src/Geometry/CMakeFiles/Geometry.dir/Tetrahedron.o
[ 17%] Building CXX object src/Geometry/CMakeFiles/Geometry.dir/TetrahedronElementField.o
[ 17%] Building CXX object src/models/CMakeFiles/models.dir/TriangleEdgeFromNodeModel.o
[ 17%] Building CXX object src/Geometry/CMakeFiles/Geometry.dir/GeometryStream.o
[ 18%] Building CXX object src/models/CMakeFiles/models.dir/TetrahedronEdgeFromNodeModel.o
Linking CXX static library libGeometry.a
[ 18%] Built target Geometry
[ 18%] Building CXX object src/models/CMakeFiles/models.dir/VectorTriangleEdgeModel.o
Scanning dependencies of target math
[ 19%] Building CXX object src/math/CMakeFiles/math.dir/TimeData.o
[ 19%] Building CXX object src/math/CMakeFiles/math.dir/DenseMatrix.o
[ 20%] Building CXX object src/models/CMakeFiles/models.dir/VectorTetrahedronEdgeModel.o
[ 21%] Building CXX object src/math/CMakeFiles/math.dir/SuperLUPreconditioner.o
/install/DevSim/devsim-master/src/math/SuperLUPreconditioner.cc:21 from included files:
/install/DevSim/devsim-master/src/math/SuperLUData.hh:27:25: error: supermatrix.h:  is not there.
[ 21%] Building CXX object src/models/CMakeFiles/models.dir/ParallelOpEqual.o
In file included from /install/DevSim/devsim-master/src/math/SuperLUPreconditioner.cc:21:
/install/DevSim/devsim-master/src/math/SuperLUData.hh:54: error: ISO C++ forbids declaration of ‘SuperMatrix’ with no type
/install/DevSim/devsim-master/src/math/SuperLUData.hh:54: error: expected ‘;’ before ‘*’ token
/install/DevSim/devsim-master/src/math/SuperLUData.hh:55: error: ISO C++ forbids declaration of ‘SuperMatrix’ with no type
/install/DevSim/devsim-master/src/math/SuperLUData.hh:55: error: expected ‘;’ before ‘*’ token
make[2]: *** [src/math/CMakeFiles/math.dir/SuperLUPreconditioner.o]  error 1
make[1]: *** [src/math/CMakeFiles/math.dir/all]  error 2
make[1]: *** waiting for incomplete jobs
Linking CXX static library libmodels.a
[ 21%] Built target models
make: *** [all] error2

juanesanchez, 2015-08-12:
When diagnosing build problems, please use
make -j1

Please follow the instructions in the INSTALL file to download and install
SuperLU and ACML.

Juan

chihiro0019, 2015-08-12:
I found several things, but not yet completed.

I needed to use redhat here.
5. sh ../superlu_ubuntu_12.04.sh	superlu_redhat_6.5.sh

This was done

I needed to install bison

yum install bison bison-devel

This was done, then,

3. make -j2

[root@JPC20320148 linux_x86_64_release]# pwd
/install/DevSim/devsim-master/linux_x86_64_release
[root@JPC20320148 linux_x86_64_release]# make -j2
[ 14%] [ 19%] Built target Geometry
Built target models
[ 24%] [ 32%] Built target modelcomp
Built target math
[ 34%] [ 34%] Built target Equation
[ 35%] Building CXX object src/AutoEquation/CMakeFiles/AutoEquation.dir/ModelExprEval.o
Building CXX object src/AutoEquation/CMakeFiles/AutoEquation.dir/ModelExprData.o
/install/DevSim/devsim-master/src/AutoEquation/ModelExprEval.cc:43:24: error: EngineAPI.hh: そのようなファイルやディレクトリはありません
/install/DevSim/devsim-master/src/AutoEquation/ModelExprEval.cc: In member function ‘MEE::ModelExprData MEE::ModelExprEval::EvaluateModelType(Eqo::EqObjPtr)’:
/install/DevSim/devsim-master/src/AutoEquation/ModelExprEval.cc:102: error: ‘EngineAPI’ has not been declared
/install/DevSim/devsim-master/src/AutoEquation/ModelExprEval.cc: In member function ‘MEE::ModelExprData MEE::ModelExprEval::EvaluateVariableType(Eqo::EqObjPtr)’:
/install/DevSim/devsim-master/src/AutoEquation/ModelExprEval.cc:223: error: ‘EngineAPI’ has not been declared
/install/DevSim/devsim-master/src/AutoEquation/ModelExprEval.cc: In member function ‘MEE::ModelExprData MEE::ModelExprEval::EvaluateProductType(Eqo::EqObjPtr)’:
/install/DevSim/devsim-master/src/AutoEquation/ModelExprEval.cc:268: error: ‘EngineAPI’ has not been declared
/install/DevSim/devsim-master/src/AutoEquation/ModelExprEval.cc: In member function ‘MEE::ModelExprData MEE::ModelExprEval::EvaluateIfType(Eqo::EqObjPtr)’:
/install/DevSim/devsim-master/src/AutoEquation/ModelExprEval.cc:299: error: ‘EngineAPI’ has not been declared
/install/DevSim/devsim-master/src/AutoEquation/ModelExprEval.cc: In member function ‘MEE::ModelExprData MEE::ModelExprEval::EvaluateIfElseType(Eqo::EqObjPtr)’:
/install/DevSim/devsim-master/src/AutoEquation/ModelExprEval.cc:330: error: ‘EngineAPI’ has not been declared
/install/DevSim/devsim-master/src/AutoEquation/ModelExprEval.cc: In member function ‘MEE::ModelExprData MEE::ModelExprEval::EvaluateAddType(Eqo::EqObjPtr)’:
/install/DevSim/devsim-master/src/AutoEquation/ModelExprEval.cc:365: error: ‘EngineAPI’ has not been declared
/install/DevSim/devsim-master/src/AutoEquation/ModelExprEval.cc: In member function ‘MEE::ModelExprData MEE::ModelExprEval::EvaluateFunctionType(Eqo::EqObjPtr)’:
/install/DevSim/devsim-master/src/AutoEquation/ModelExprEval.cc:380: error: ‘EngineAPI’ has not been declared
/install/DevSim/devsim-master/src/AutoEquation/ModelExprEval.cc:380: error: expected initializer before ‘etype’
/install/DevSim/devsim-master/src/AutoEquation/ModelExprEval.cc:383: error: ‘EngineAPI’ has not been declared
/install/DevSim/devsim-master/src/AutoEquation/ModelExprEval.cc:386: error: ‘EngineAPI’ has not been declared
/install/DevSim/devsim-master/src/AutoEquation/ModelExprEval.cc: In member function ‘MEE::ModelExprData MEE::ModelExprEval::EvaluateInvalidType(Eqo::EqObjPtr)’:
/install/DevSim/devsim-master/src/AutoEquation/ModelExprEval.cc:405: error: ‘EngineAPI’ has not been declared
/install/DevSim/devsim-master/src/AutoEquation/ModelExprEval.cc: In member function ‘MEE::ModelExprData MEE::ModelExprEval::EvaluateConstantType(Eqo::EqObjPtr)’:
/install/DevSim/devsim-master/src/AutoEquation/ModelExprEval.cc:414: error: ‘EngineAPI’ was not declared in this scope
/install/DevSim/devsim-master/src/AutoEquation/ModelExprEval.cc: In member function ‘MEE::ModelExprData MEE::ModelExprEval::eval_function(Eqo::EqObjPtr)’:
/install/DevSim/devsim-master/src/AutoEquation/ModelExprEval.cc:422: error: ‘EngineAPI’ has not been declared
/install/DevSim/devsim-master/src/AutoEquation/ModelExprEval.cc:422: error: expected initializer before ‘etype’
/install/DevSim/devsim-master/src/AutoEquation/ModelExprEval.cc:438: error: ‘EngineAPI’ has not been declared
/install/DevSim/devsim-master/src/AutoEquation/ModelExprEval.cc:446: error: ‘EngineAPI’ has not been declared
/install/DevSim/devsim-master/src/AutoEquation/ModelExprEval.cc:449: error: ‘EngineAPI’ has not been declared
/install/DevSim/devsim-master/src/AutoEquation/ModelExprEval.cc:452: error: ‘EngineAPI’ has not been declared
/install/DevSim/devsim-master/src/AutoEquation/ModelExprEval.cc:455: error: ‘EngineAPI’ has not been declared
/install/DevSim/devsim-master/src/AutoEquation/ModelExprEval.cc:458: error: ‘EngineAPI’ has not been declared
/install/DevSim/devsim-master/src/AutoEquation/ModelExprEval.cc:461: error: ‘EngineAPI’ has not been declared
/install/DevSim/devsim-master/src/AutoEquation/ModelExprEval.cc:464: error: ‘EngineAPI’ has not been declared
/install/DevSim/devsim-master/src/AutoEquation/ModelExprEval.cc:467: error: ‘EngineAPI’ has not been declared
/install/DevSim/devsim-master/src/AutoEquation/ModelExprEval.cc:468: error: ‘EngineAPI’ has not been declared
/install/DevSim/devsim-master/src/AutoEquation/ModelExprEval.cc:469: error: ‘EngineAPI’ has not been declared
/install/DevSim/devsim-master/src/AutoEquation/ModelExprEval.cc:470: error: ‘EngineAPI’ has not been declared
/install/DevSim/devsim-master/src/AutoEquation/ModelExprEval.cc:471: error: ‘EngineAPI’ has not been declared
[... cut ...]
/install/DevSim/devsim-master/src/AutoEquation/ModelExprEval.cc:472: error: ‘EngineAPI’ has not been declared
/install/DevSim/devsim-master/src/AutoEquation/ModelExprEval.cc:485: error: ‘EngineAPI’ has not been declared
/install/DevSim/devsim-master/src/AutoEquation/ModelExprEval.cc:492: error: ‘EngineAPI’ has not been declared
make[2]: *** [src/AutoEquation/CMakeFiles/AutoEquation.dir/ModelExprEval.o]  error 1
make[1]: *** [src/AutoEquation/CMakeFiles/AutoEquation.dir/all]  error 2
make: *** [all]  eror 2
[root@JPC20320148 linux_x86_64_release]#

juanesanchez, 2015-08-12:
EngineAPI.hh should be in
external/symdiff/include

make sure that symdiff built correctly, and the file should be put there 
automatically.

For more debugging information:
make VERBOSE=1

chihiro0019, 2015-08-12:
--- Thank you for your reply.  I have external/symdiff/include as below

# pwd
/install/DevSim/devsim-master/external/symdiff/include
# ls
.  ..  EngineAPI.hh  import.hh  parseexpr.hh

# pwd
/install/DevSim/devsim-master/external/symdiff/linux_x86_64_release
# make VERBOSE=1
...... no problem .....

--- while make VERBOSE=1, here are 3 problems in the log
/install/DevSim/devsim-master/src/Data/MaterialDB.cc:23:21: error: sqlite3.h:  No such file or directory
error 1
error 2

--- log of make VERBOSE=1
# pwd
/install/DevSim/devsim-master/linux_x86_64_release
# make VERBOSE=1
/usr/bin/cmake -H/install/DevSim/devsim-master -B/install/DevSim/devsim-master/linux_x86_64_release --check-build-system CMakeFiles/Makefile.cmake 0
/usr/bin/cmake -E cmake_progress_start /install/DevSim/devsim-master/linux_x86_64_release/CMakeFiles /install/DevSim/devsim-master/linux_x86_64_release/CMakeFiles/progress.marks
make -f CMakeFiles/Makefile2 all
make[1]: ディレクトリ `/install/DevSim/devsim-master/linux_x86_64_release' に入ります
make -f src/Geometry/CMakeFiles/Geometry.dir/build.make src/Geometry/CMakeFiles/Geometry.dir/depend
make[2]: ディレクトリ `/install/DevSim/devsim-master/linux_x86_64_release' に入ります
cd /install/DevSim/devsim-master/linux_x86_64_release && /usr/bin/cmake -E cmake_depends "Unix Makefiles" /install/DevSim/devsim-master /install/DevSim/devsim-master/src/Geometry /install/DevSim/devsim-master/linux_x86_64_release /install/DevSim/devsim-master/linux_x86_64_release/src/Geometry /install/DevSim/devsim-master/linux_x86_64_release/src/Geometry/CMakeFiles/Geometry.dir/DependInfo.cmake --color=
make[2]: ディレクトリ `/install/DevSim/devsim-master/linux_x86_64_release' から出ます
make -f src/Geometry/CMakeFiles/Geometry.dir/build.make src/Geometry/CMakeFiles/Geometry.dir/build
make[2]: ディレクトリ `/install/DevSim/devsim-master/linux_x86_64_release' に入ります
make[2]: `src/Geometry/CMakeFiles/Geometry.dir/build' に対して行うべき事はありません.
make[2]: ディレクトリ `/install/DevSim/devsim-master/linux_x86_64_release' から出ます
/usr/bin/cmake -E cmake_progress_report /install/DevSim/devsim-master/linux_x86_64_release/CMakeFiles  22 23 24 25 26 27 28
[  7%] Built target Geometry
make -f src/models/CMakeFiles/models.dir/build.make src/models/CMakeFiles/models.dir/depend
make[2]: ディレクトリ `/install/DevSim/devsim-master/linux_x86_64_release' に入ります
cd /install/DevSim/devsim-master/linux_x86_64_release && /usr/bin/cmake -E cmake_depends "Unix Makefiles" /install/DevSim/devsim-master /install/DevSim/devsim-master/src/models /install/DevSim/devsim-master/linux_x86_64_release /install/DevSim/devsim-master/linux_x86_64_release/src/models /install/DevSim/devsim-master/linux_x86_64_release/src/models/CMakeFiles/models.dir/DependInfo.cmake --color=
make[2]: ディレクトリ `/install/DevSim/devsim-master/linux_x86_64_release' から出ます
make -f src/models/CMakeFiles/models.dir/build.make src/models/CMakeFiles/models.dir/build
make[2]: ディレクトリ `/install/DevSim/devsim-master/linux_x86_64_release' に入ります
make[2]: `src/models/CMakeFiles/models.dir/build' に対して行うべき事はありません.
make[2]: ディレクトリ `/install/DevSim/devsim-master/linux_x86_64_release' から出ます
/usr/bin/cmake -E cmake_progress_report /install/DevSim/devsim-master/linux_x86_64_release/CMakeFiles  73 74 75 76 77 78 79 80 81 82 83 84
[ 19%] Built target models
make -f src/math/CMakeFiles/math.dir/build.make src/math/CMakeFiles/math.dir/depend
make[2]: ディレクトリ `/install/DevSim/devsim-master/linux_x86_64_release' に入ります
cd /install/DevSim/devsim-master/linux_x86_64_release && /usr/bin/cmake -E cmake_depends "Unix Makefiles" /install/DevSim/devsim-master /install/DevSim/devsim-master/src/math /install/DevSim/devsim-master/linux_x86_64_release /install/DevSim/devsim-master/linux_x86_64_release/src/math /install/DevSim/devsim-master/linux_x86_64_release/src/math/CMakeFiles/math.dir/DependInfo.cmake --color=
make[2]: ディレクトリ `/install/DevSim/devsim-master/linux_x86_64_release' から出ます
make -f src/math/CMakeFiles/math.dir/build.make src/math/CMakeFiles/math.dir/build
make[2]: ディレクトリ `/install/DevSim/devsim-master/linux_x86_64_release' に入ります
make[2]: `src/math/CMakeFiles/math.dir/build' に対して行うべき事はありません.
make[2]: ディレクトリ `/install/DevSim/devsim-master/linux_x86_64_release' から出ます
/usr/bin/cmake -E cmake_progress_report /install/DevSim/devsim-master/linux_x86_64_release/CMakeFiles  46 47 48 49 50 51 52 53
[ 27%] Built target math
make -f src/adiff/CMakeFiles/modelcomp.dir/build.make src/adiff/CMakeFiles/modelcomp.dir/depend
make[2]: ディレクトリ `/install/DevSim/devsim-master/linux_x86_64_release' に入ります
cd /install/DevSim/devsim-master/linux_x86_64_release && /usr/bin/cmake -E cmake_depends "Unix Makefiles" /install/DevSim/devsim-master /install/DevSim/devsim-master/src/adiff /install/DevSim/devsim-master/linux_x86_64_release /install/DevSim/devsim-master/linux_x86_64_release/src/adiff /install/DevSim/devsim-master/linux_x86_64_release/src/adiff/CMakeFiles/modelcomp.dir/DependInfo.cmake --color=
make[2]: ディレクトリ `/install/DevSim/devsim-master/linux_x86_64_release' から出ます
make -f src/adiff/CMakeFiles/modelcomp.dir/build.make src/adiff/CMakeFiles/modelcomp.dir/build
make[2]: ディレクトリ `/install/DevSim/devsim-master/linux_x86_64_release' に入ります
make[2]: `src/adiff/CMakeFiles/modelcomp.dir/build' に対して行うべき事はありません.
make[2]: ディレクトリ `/install/DevSim/devsim-master/linux_x86_64_release' から出ます
/usr/bin/cmake -E cmake_progress_report /install/DevSim/devsim-master/linux_x86_64_release/CMakeFiles  68 69 70 71 72
[ 32%] Built target modelcomp
[... cut ...]
[... 135 lines cut ...]
  description_(o.description_),
  value_(o.value_),
  entry_type_(o.entry_type_)
{
}

MaterialDBEntry &MaterialDBEntry::operator=(const MaterialDBEntry &o)
{
  if (this != &o)
  {
    unit_ = o.unit_;
    description_ = o.description_;
    value_ = o.value_;
    entry_type_ = o.entry_type_;
  }
  return *this;
}

MaterialDBEntry::MaterialDBEntry(const std::string &u, const std::string &d, const ObjectHolder &v, EntryType_t et) :
  unit_(u),
  description_(d),
  value_(v),
  entry_type_(et)
{
}

juanesanchez, 2015-08-13:
Please install:
sqlite
sqlite-devel

chihiro0019, 2015-08-13:
Thank you, probably the last problem to make -j2 is about gfortran64.
gcc-gfortran.x86_64 0:4.4.7-16.el6 was installed, but,

$ pwd
/install/DevSim/devsim-master/linux_x86_64_release

$ make -j2                                                                              
[ 12%] Built target models
...
...
[ 99%] Built target pythonapi_interpreter
make[2]: *** `src/main/devsim_tcl'  needs the target `/opt/acml5.3.0/gfortran64/lib/libacml.a'  but, there is no rule to make it
make[2]: *** `src/main/devsim_py'  needs the target  `/opt/acml5.3.0/gfortran64/lib/libacml.a'  but, there is no rule to make it
...

juanesanchez, 2015-08-13:
Please download and install

acml-5-3-0-gfortran-64bit.tgz 
<http://developer.amd.com/tools-and-sdks/archive/amd-core-math-library-acml/acml-archive-downloads/#>

from
http://developer.amd.com/tools-and-sdks/archive/amd-core-math-library-acml/acml-archive-downloads/

chihiro0019, 2015-08-13:
In stalling directory is   /install/DevSim/devsim-master/src  ?

$ pwd
/install/DevSim/devsim-master/src
$ ls
.               Circuit     MathEval                       common_api                              main      pythonapi
..              Data        README.64-bit                  contents-acml-5-3-0-gfortran-64bit.tgz  math      tclapi
ACML-EULA.txt   Equation    acml-5-3-0-gfortran-64bit.tgz  doxyfile                                meshing   utility
AutoEquation    GeomModels  adiff                          errorSystem                             models
CMakeLists.txt  Geometry    commands 

 $ pwd
/install/DevSim/devsim-master/linux_x86_64_release
$ make -j2
...
[ 99%] make[2]: *** `src/main/devsim_tcl'  needs the target `/opt/acml5.3.0/gfortran64/lib/libacml.a'  there is no rule to  make it.

juanesanchez, 2015-08-13:
It needs to be put installed in:
/opt/acml5.3.0
If this is not possible, you need to change the path for BLAS_ARCHIVE in 
redhat_6.5.cmake.

juanesanchez, 2015-08-13:
mkdir tmp
cd tmp
tar xzvf ../acml-5-3-0-gfortran-64bit.tgz
bash install-acml-5-3-0-gfortran-64bit.sh

chihiro0019, 2015-08-13:
Thank you, I could compile DevSim. But I met some errors at ctest -j2.
I tried -j1, too, but the result was the same.

$ ctest -j2
Test project /install/DevSim/devsim-master/linux_x86_64_release
      Start  1: testing/cap2
      Start  2: testing/equation1
...
...
Total Test time (real) =  96.15 sec

The following tests FAILED:
         37 - testing/mos_2d_restart2 (Failed)
         49 - testing/sqlite1 (Failed)
         52 - testing/thread_1 (Failed)
         54 - testing/fpetest2 (Timeout)
Errors while running CTest

juanesanchez, 2015-08-13:
I am glad you got the program to compile.  Unfortunately the tests will 
only be 100% successful on an older amd machine running ubuntu 12.04.  
This is because the tests are prone to numerical differences with the 
compiler and the cpu.

A simple test is:
cd examples/capacitance
../../linux_x86_64_release/src/main/devsim_py  cap1d.py

If you see a result like this:
contact: contact1 charge: 3.45150e-13
contact: contact2 charge: -3.45150e-13

That is good.

You can also look at the following file to see what the file differences 
are:
./linux_x86_64_release/Testing/Temporary/LastTest.log
./linux_x86_64_release/Testing/Temporary/LastTestsFailed.log

chihiro0019, 2015-08-13:
Thank you very much, it seems I could complete.
I'm going to run my test data, then I might ask you again about how to use.
Thank you again for your kind support, good bye for the time being.

closed 2015-08-13 by chihiro0019