devsim/devsim #10: About given testing

chihiro0019, 2015-08-17:
Dear,  juanesanchez-san.

Since I am a novice of DevSim, I can't analyze what's wrong with the given testing.
Please advice. Seems sqlite3.something is missing.

 [root@JPC20320148 testing]# ls sql
sqlite1.out  sqlite1.tcl  sqlite2.out  sqlite2.py   sqlite3.out  sqlite3.py   
[root@JPC20320148 testing]# ls sqlite1
sqlite1.out  sqlite1.tcl  
[root@JPC20320148 testing]# ls sqlite1.tcl 
sqlite1.tcl
[root@JPC20320148 testing]# emacs sqlite1.tcl
[1] 8797
[root@JPC20320148 testing]# ls sql
sqlite1.out  sqlite1.tcl  sqlite2.out  sqlite2.py   sqlite3.out  sqlite3.py   
[root@JPC20320148 testing]# ls sqlite3
ls: cannot access sqlite3:    doesn't exist.
[root@JPC20320148 testing]# ls sqli*
sqlite1.out  sqlite1.tcl  sqlite2.out  sqlite2.py  sqlite3.out  sqlite3.py
[root@JPC20320148 testing]# pwd
/install/DevSim/devsim-master/testing
[root@JPC20320148 testing]# ls sqli*
sqlite1.out  sqlite1.tcl  sqlite2.out  sqlite2.py  sqlite3.out  sqlite3.py
[root@JPC20320148 testing]# devsim_tcl sqlite1.tcl

----------------------------------------

 DEVSIM

 Version: Beta 0.01

 Copyright 2009-2014 Devsim LLC

----------------------------------------

can't find package sqlite3
    while executing
"package require sqlite3"
    (file "sqlite1.tcl" line 15)

juanesanchez, 2015-08-17:
Hi,

Please install:
yum install sqlite-tcl

chihiro0019, 2015-08-17:
Thanks, it was solved.

closed 2015-08-17 by chihiro0019