devsim/devsim #11: from python_packages.simple_physics import *

chihiro0019, 2015-08-17:
Dear, juanesanchez-san.   Sorry to bother you many times.

$ pwd
/install/DevSim/devsim-master/examples/diode
$ devsim_py diode_2d.py

------------ diode_2d.py ----------------------------
from ds import *
from python_packages.simple_physics import *
----------------------------------------------------------

This looks working. However with another test case placed in a different directory.
$pwd
/Application/DevSim/Holor/1stFromSony/devsim_input

------------- input.py ----------------------------------
from python_packages.simple_physics import *
from python_packages.ramp import *
-----------------------------------------------------------

$ devsim_py input.py

----------------------------------- error message ----------
Traceback (most recent call last):
  File "input.py", line 4, in <module>
    from python_packages.simple_physics import *
ImportError: No module named python_packages.simple_physics
-------------------------------------------------------------------

DevSim didn't understand from description. How can I tell DevSim where
is python_packages.simple_physics ?

chihiro0019, 2015-08-17:
Now I understand. Simply creating a symbolic link works well.  Thanks.

juanesanchez, 2015-08-18:
Symbolic linking works.  You can also set PYTHONPATH.

In bash:
export PYTHONPATH=/install/DevSim/devsim-master

chihiro0019, 2015-08-19:
Thank you.
$ setenv PYTHONPATH /install/DevSim/devsim-master
worked.

chihiro0019, 2015-08-19:
Could you explain how the following statement works ?

from ds import *

Does this link ds to /install/DevSim/devsim-master/linux_x86_64_release/src/pythonapi ?
ds seems to stand for DevSim, but I don't know where actually it is.

juanesanchez, 2015-08-20:
devsim has a built in python interpreter.  All of the commands are in the
ds module which is loaded into devsim at startup.  If you want to use the
commands directly, without the ds prefix, you use that command to make them
available.

chihiro0019, 2015-08-20:
Thank you, it means ds is built in DevSim module itself. I understand.

closed 2015-08-20 by chihiro0019