devsim/devsim #13: Installation on Windows

RineJT, 2015-08-20:
I want to install devsim directly in Windows 7 (and not in Cygwin or a vm running Linux). Since there is no Windows binary on the GitHub binaries page, I attempted to install from source. The installation did put directories and files in C:\Program Files\devsim, however, no binaries were generated.
Here are the steps I took:
I downloaded the source from https://github.com/devsim/devsim/archive/master.zip. After unzipping it, I read the Windows.txt file.
Next, I followed its instructions. I installed ActiveTcl, ActivePython, and the Visual C++ redistributable. Next, I set the following paths in path C:\Python27;C:\Tcl\bin;C:\devsim. Looking in the unzipped directory, I observed the Win32.cmake file. I assume Cmake is like make and so I downloaded, installed and then opened the Cmake GUI on the Windows desktop and entered the path to the source and the path to the binaries (C:\Program Files\devsim). I clicked on configure and received the following configuration  error: "CMake Error at CMakeLists.txt:5 (INCLUDE):  include could not find load file: win32.cmake.cmake". Looking at the CMakeLists.txt, a particular .cmake file is referenced through the ${DEVSIM_CONFIG}.cmake variable to I set this to Win32.cmake. Afterwards, I no longer received the error (but did receive a few warnings). I proceded to generate the build (Cmake build button). The project was built successfully, at least no errors, however, no devsim executable was generated.
How do I get the devsim executable built?
In doc/devsim.pdf, the other dependencies are listed, I assume these must also be installed in Windows:
BLAS
LAPACK
CGNS
SQLite3
SuperLU
zlib
Is this correct?

juanesanchez, 2015-08-20:
There is a second package with instructions for installing the dependencies.
https://github.com/devsim/devsim_win32_deps

It contains instructions to download and compile the dependencies in the
README.txt

RineJT, 2015-08-20:
Juan,
Thanks very much!
I’ll give it a try.

John Rine

https://github.com/devsim/devsim_win32_deps

RineJT, 2015-08-28:
Juan,
I had success building DEVSIM and its dependents on a 32-bit Windows system using the instructions you provided. Thank you very much!
However, we would rather have it on a different computer so I am now installing it on a 64-bit system instead. Everything was going well until I got to the SymDiff Build. I get two errors (see Cygwin console output below). Do you have any idea what is causing it and how can it be fixed?

John Rine
rinejt@SPLBSBS052 /cygdrive/c/DEVSIM/symdiff-master/win64
$ "$CMAKE" --build . --config Release -- /m:1
Microsoft (R) Build Engine version 12.0.31101.0
[Microsoft .NET Framework, version 4.0.30319.18408]
Copyright (C) Microsoft Corporation. All rights reserved.

Build started 8/28/2015 1:54:51 PM.
Project "C:\DEVSIM\symdiff-master\win64\ALL_BUILD.vcxproj" on node 1 (default ta                                     rgets).
Project "C:\DEVSIM\symdiff-master\win64\ALL_BUILD.vcxproj" (1) is building "C:\D                                     EVSIM\symdiff-master\win64\ZERO_CHECK.vcxproj" (2) on node 1 (default targets).
PrepareForBuild:
  Creating directory "x64\Release\ZERO_CHECK\".
  Creating directory "x64\Release\ZERO_CHECK\ZERO_CHECK.tlog\".
InitializeBuildStatus:
  Creating "x64\Release\ZERO_CHECK\ZERO_CHECK.tlog\unsuccessfulbuild" because "A                                     lwaysCreate" was specified.
CustomBuild:
  Checking Build System
  CMake does not need to re-run because C:/DEVSIM/symdiff-master/win64/CMakeFile                                     s/generate.stamp is up-to-date.
  CMake does not need to re-run because C:/DEVSIM/symdiff-master/win64/src/CMake                                     Files/generate.stamp is up-to-date.
  CMake does not need to re-run because C:/DEVSIM/symdiff-master/win64/src/engin                                     e/CMakeFiles/generate.stamp is up-to-date.
  CMake does not need to re-run because C:/DEVSIM/symdiff-master/win64/src/tclco                                     mp/CMakeFiles/generate.stamp is up-to-date.
  CMake does not need to re-run because C:/DEVSIM/symdiff-master/win64/src/pycom                                     p/CMakeFiles/generate.stamp is up-to-date.
  CMake does not need to re-run because C:/DEVSIM/symdiff-master/win64/testing/C                                     MakeFiles/generate.stamp is up-to-date.
FinalizeBuildStatus:
  Deleting file "x64\Release\ZERO_CHECK\ZERO_CHECK.tlog\unsuccessfulbuild".
  Touching "x64\Release\ZERO_CHECK\ZERO_CHECK.tlog\ZERO_CHECK.lastbuildstate".
Done Building Project "C:\DEVSIM\symdiff-master\win64\ZERO_CHECK.vcxproj" (defau                                     lt targets).
Project "C:\DEVSIM\symdiff-master\win64\ALL_BUILD.vcxproj" (1) is building "C:\D                                     EVSIM\symdiff-master\win64\src\engine\symdiff.vcxproj" (3) on node 1 (default ta                                     rgets).
PrepareForBuild:
  Creating directory "symdiff.dir\Release\".
  Creating directory "C:\DEVSIM\symdiff-master\win64\src\engine\Release\".
  Creating directory "symdiff.dir\Release\symdiff.tlog\".
InitializeBuildStatus:
  Creating "symdiff.dir\Release\symdiff.tlog\unsuccessfulbuild" because "AlwaysC                                     reate" was specified.
CustomBuild:
  Building Custom Rule C:/DEVSIM/symdiff-master/src/engine/CMakeLists.txt
  CMake does not need to re-run because C:\DEVSIM\symdiff-master\win64\src\engin                                     e\CMakeFiles\generate.stamp is up-to-date.
  Generating mcParser.cc, mcParser.hh
  The system cannot find the path specified.
C:\Program Files (x86)\MSBuild\Microsoft.Cpp\v4.0\V120\Microsoft.CppCommon.targe                                     ts(170,5): error MSB6006: "cmd.exe" exited with code 3. [C:\DEVSIM\symdiff-maste                                     r\win64\src\engine\symdiff.vcxproj]
Done Building Project "C:\DEVSIM\symdiff-master\win64\src\engine\symdiff.vcxproj                                     " (default targets) -- FAILED.
Project "C:\DEVSIM\symdiff-master\win64\ALL_BUILD.vcxproj" (1) is building "C:\D                                     EVSIM\symdiff-master\win64\src\engine\symdiff_static.vcxproj" (5) on node 1 (def                                     ault targets).
PrepareForBuild:
  Creating directory "symdiff_static.dir\Release\".
  Creating directory "symdiff_static.dir\Release\symdiff_static.tlog\".
InitializeBuildStatus:
  Creating "symdiff_static.dir\Release\symdiff_static.tlog\unsuccessfulbuild" be                                     cause "AlwaysCreate" was specified.
CustomBuild:
  Building Custom Rule C:/DEVSIM/symdiff-master/src/engine/CMakeLists.txt
  CMake does not need to re-run because C:\DEVSIM\symdiff-master\win64\src\engin                                     e\CMakeFiles\generate.stamp is up-to-date.
  Generating mcParser.cc, mcParser.hh
  The system cannot find the path specified.
C:\Program Files (x86)\MSBuild\Microsoft.Cpp\v4.0\V120\Microsoft.CppCommon.targe                                     ts(170,5): error MSB6006: "cmd.exe" exited with code 3. [C:\DEVSIM\symdiff-maste                                     r\win64\src\engine\symdiff_static.vcxproj]
Done Building Project "C:\DEVSIM\symdiff-master\win64\src\engine\symdiff_static.                                     vcxproj" (default targets) -- FAILED.
Done Building Project "C:\DEVSIM\symdiff-master\win64\ALL_BUILD.vcxproj" (defaul                                     t targets) -- FAILED.

Build FAILED.

[... cut ...]
"C:\DEVSIM\symdiff-master\win64\src\engine\symdiff_static.vcxproj" (default targ                                     et) (5) ->
  C:\Program Files (x86)\MSBuild\Microsoft.Cpp\v4.0\V120\Microsoft.CppCommon.tar                                     gets(170,5): error MSB6006: "cmd.exe" exited with code 3. [C:\DEVSIM\symdiff-mas                                     ter\win64\src\engine\symdiff_static.vcxproj]

    0 Warning(s)
    2 Error(s)

Time Elapsed 00:00:01.07

https://github.com/devsim/devsim_win32_deps

juanesanchez, 2015-08-28:
Hi John,

Glad to hear you can build the software.

It has been a while since I've done the Windows build.  You may want to 
see if there is a log file which shows what it was trying to do.  On 
other platforms, the VERBOSE macro may be defined to output more 
information.

This seems to indicate that flex and bison may not be installed in cygwin.
 > Generating mcParser.cc, mcParser.hh
 > The system cannot find the path specified.

Please look in the WIN32 section of symdiff/CMakeLists.txt and 
win32.cmake to make sure all the binaries you need exist.

Note that the Python and Tcl paths I assume may be non standard for a 64 
bit build.

Juan

RineJT, 2015-08-28:
Juan,
Yes, thanks very much. You are correct. In SymDiff CMakeLists.txt sets the path to Flex and  Bison as Cygwin. I have the 64-bit version installed it is Cygwin64. I changed these paths. Now it appears there is a new problem.

Thanks,

John Rine

juanesanchez, 2015-08-28:
Hi John,

I've only used 32 bit cygwin.  Since we are not linking against any 
cygwin libraries, you can use whichever version you want.

juanesanchez, 2015-10-16:
A windows 64 bit binary release is available under the releases tab.
https://github.com/devsim/devsim/releases/latest

New binary releases for Linux, Mac OS X, and Windows will be done on a more regular basis.

closed 2015-10-16 by juanesanchez