devsim/devsim #14: issue with EngineAPI.hh and ModelExprData.cc on OS X 10.10

d679, 2015-08-27:
Hi, 
I am having a problem building devsim. I am getting the following errrors, and was wondering if you have some suggestions?

[  6%] Built target Geometry
[ 17%] Built target models
[ 27%] Built target modelcomp
[ 29%] Built target math
[ 31%] Built target Equation
[ 31%] Building CXX object src/AutoEquation/CMakeFiles/AutoEquation.dir/ModelExprEval.o
[ 31%] Building CXX object src/AutoEquation/CMakeFiles/AutoEquation.dir/ModelExprData.o
/Users/sonia/Documents/Simulations/devsim/src/AutoEquation/ModelExprData.cc:4:1: error: 
      expected unqualified-id
<!DOCTYPE html>
^
/Users/sonia/Documents/Simulations/devsim/src/AutoEquation/ModelExprData.cc:13:46: error: 
      non-ASCII characters are not allowed outside of literals and identifiers
    <title>devsim/ModelExprData.cc at master · devsim/devsim · GitHub</title>
                                             ^~
/Users/sonia/Documents/Simulations/devsim/src/AutoEquation/ModelExprData.cc:13:63: error: 
      non-ASCII characters are not allowed outside of literals and identifiers
    <title>devsim/ModelExprData.cc at master · devsim/devsim · GitHub</title>
                                                             ^~
/Users/sonia/Documents/Simulations/devsim/src/AutoEquation/ModelExprData.cc:50:30: warning: 
      missing terminating ' character [-Winvalid-pp-token]
    <meta content="authenticity_token" n...
                             ^
/Users/sonia/Documents/Simulations/devsim/src/AutoEquation/ModelExprData.cc:95:32: warning: 
      missing terminating ' character [-Winvalid-pp-token]
      <form accept-charset="UTF-8" actio...
                               ^
/Users/sonia/Documents/Simulations/devsim/src/AutoEquation/ModelExprData.cc:284:754: error: 
      C++ requires a type specifier for all declarations
  ...protocol_type=clone" class="inline-form js-clone-selector-form " data-fo...
     ^~~~~~~~~~~~~
/Users/sonia/Documents/Simulations/devsim/src/AutoEquation/ModelExprData.cc:284:768: error: 
      use of undeclared identifier 'clone'
  ...clone" class="inline-form js-clone-selector-form " data-form-nonce="90d1...
     ^
/Users/sonia/Documents/Simulations/devsim/src/AutoEquation/ModelExprData.cc:284:932: error: 
      unknown type name 'padding'
  ...data-remote="true" method="post"><div style="margin:0;padding:0;display:...
                                                           ^
/Users/sonia/Documents/Simulations/devsim/src/AutoEquation/ModelExprData.cc:284:939: error: 
      expected unqualified-id
  ...data-remote="true" method="post"><div style="margin:0;padding:0;display:...
                                                                  ^
/Users/sonia/Documents/Simulations/devsim/src/AutoEquation/ModelExprData.cc:284:942: error: 
      unknown type name 'display'
  ...data-remote="true" method="post"><div style="margin:0;padding:0;display:...
                                                                     ^
/Users/sonia/Documents/Simulations/devsim/src/AutoEquation/ModelExprData.cc:284:949: error: 
      expected unqualified-id
  ...method="post"><div style="margin:0;padding:0;display:inline"><input name...
                                                         ^
/Users/sonia/Documents/Simulations/devsim/src/AutoEquation/ModelExprData.cc:284:1006: error: 
      expected unqualified-id
  ...name="utf8" type="hidden" value="&#x2713;" /><input name="authenticity_t...
                                              ^
/Users/sonia/Documents/Simulations/devsim/src/AutoEquation/ModelExprData.cc:284:1244: warning: 
      missing terminating '"' character [-Winvalid-pp-token]
  ...js-clone-selector" data-protocol="subversion" type="submit">Subversion</...
                                                               ^
/Users/sonia/Documents/Simulations/devsim/src/AutoEquation/ModelExprData.cc:408:3: error: 
      expected unqualified-id
  </div>
  ^
/Users/sonia/Documents/Simulations/devsim/src/AutoEquation/ModelExprData.cc:530:151: error: 
      expected unqualified-id
  ...<span class="pl-s"><span class="pl-pds">&quot;</span>ModelExprData.hh<sp...
                                                   ^
[... 125 lines cut ...]
  ...class="pl-pds">&lt;</span>string<span class="pl-pds">&gt;</span></span><...
                                                              ^
/Users/sonia/Documents/Simulations/devsim/src/AutoEquation/EngineAPI.hh:551:149: error: 
      expected unqualified-id
  ...<span class="pl-s"><span class="pl-pds">&lt;</span>vector<span class="pl...
                                                 ^
/Users/sonia/Documents/Simulations/devsim/src/AutoEquation/EngineAPI.hh:551:187: error: 
      expected unqualified-id
  ...class="pl-pds">&lt;</span>vector<span class="pl-pds">&gt;</span></span><...
                                                              ^
/Users/sonia/Documents/Simulations/devsim/src/AutoEquation/EngineAPI.hh:555:149: error: 
      expected unqualified-id
  ...<span class="pl-s"><span class="pl-pds">&lt;</span>set<span class="pl-pd...
                                                 ^
fatal error: too many errors emitted, stopping now [-ferror-limit=]
5 warnings and 20 errors generated.
make[2]: *** [src/AutoEquation/CMakeFiles/AutoEquation.dir/ModelExprEval.o] Error 1
make[1]: *** [src/AutoEquation/CMakeFiles/AutoEquation.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 68%] Building CXX object src/MathEval/CMakeFiles/MathEval.dir/MiscMathFunc.o
[ 69%] Building CXX object src/MathEval/CMakeFiles/MathEval.dir/MathWrapper.o
[ 69%] Building CXX object src/MathEval/CMakeFiles/MathEval.dir/MathPacket.o
[ 69%] Linking CXX static library libMathEval.a
[ 69%] Built target MathEval
make: *** [all] Error 2

juanesanchez, 2015-08-27:
Hi,

The error messages look like the files are html instead of C++ source 
code.  Please make sure that the files are downloaded correctly, and 
they somehow did not become converted to a different format.

If you try:
make VERBOSE=1

this will tell the command line being invoked by the compiler and may 
help to debug issues.

d679, 2015-08-28:
Thank you, it works now.

Sonia

closed 2015-09-25 by juanesanchez