devsim/devsim #15: Making symdiff failed

chihiro0019, 2015-11-09:
Sorry to bother you, but I could not make 'symdiff' for CentOS5.1
Here I attached log messages. The errors happened at about 95%
of the entire 'make -j2' process. I had no problem until this point.
Please advice me. I'd like to solve this without upgrading to CentOS6.

                          Kimihiro

[root@gstj01 symdiff]# cd linux_x86_64_release
[root@gstj01 linux_x86_64_release]# make -j2
[  1%] [  3%] Generating mcParser.cc, mcParser.hh
Generating mcParser.cc, mcParser.hh

...

[ 93%] Built target symdiff_static
[ 93%] Built target symdiff
Scanning dependencies of target symdiff_tcl
Scanning dependencies of target symdiff_py
make[2]: *** `src/pycomp/symdiff.so'  there is no rule to make  `/root/anaconda/lib/libpython2.7.so' 
....
[ 94%] [ 96%] Building CXX object src/pycomp/CMakeFiles/symdiff_py.dir/PythonSymdiffCommands.o
Building CXX object src/tclcomp/CMakeFiles/symdiff_tcl.dir/TclCommands.o
/install/devsim-master/external/symdiff/src/pycomp/PythonSymdiffCommands.cc:23:20: error: there is no Python.h: 
/install/devsim-master/external/symdiff/src/pycomp/PythonSymdiffCommands.cc:24:20: error:  there is no import.h:
[ 98%] Building CXX object src/tclcomp/CMakeFiles/symdiff_tcl.dir/TclSymdiffCommands.o
/install/devsim-master/external/symdiff/src/pycomp/PythonSymdiffCommands.cc:38: error: expected constructor, destructor, or type conversion before ‘*’ token
/install/devsim-master/external/symdiff/src/pycomp/PythonSymdiffCommands.cc:43: error: ‘PyObject’ was not declared in this scope
/install/devsim-master/external/symdiff/src/pycomp/PythonSymdiffCommands.cc:43: error: ‘args’ was not declared in this scope
/install/devsim-master/external/symdiff/src/pycomp/PythonSymdiffCommands.cc:44: error: expected ‘,’ or ‘;’ before ‘{’ token
make[2]: *** [src/pycomp/CMakeFiles/symdiff_py.dir/PythonSymdiffCommands.o]  error 1
make[1]: *** [src/pycomp/CMakeFiles/symdiff_py.dir/all] error 2
make[1]: *** ....
make[2]: *** `src/tclcomp/libsymdifftcl.so'  there is no rule to make  `/root/anaconda/lib/libtclstub8.5.a' 
make[2]: *** ....
[100%] Building CXX object src/tclcomp/CMakeFiles/symdiff_tcl.dir/tclcomp.o
/install/devsim-master/external/symdiff/src/tclcomp/tclcomp.cc:43: error: expected constructor, destructor, or type conversion before ‘*’ token
/install/devsim-master/external/symdiff/src/tclcomp/tclcomp.cc:44: error: expected constructor, destructor, or type conversion before ‘*’ token
make[2]: *** [src/tclcomp/CMakeFiles/symdiff_tcl.dir/tclcomp.o]  error 1
make[1]: *** [src/tclcomp/CMakeFiles/symdiff_tcl.dir/all]  error 2
make: *** [all]  error 2
[root@gstj01 linux_x86_64_release]# pwd
/install/devsim-master/external/symdiff/linux_x86_64_release

juanesanchez, 2015-11-09:
You need to use the:
symdiff/scripts/setup_centos5.sh

script.  It assumes that you have Anaconda Python from Continuum 
installed in your home directory.

It is available from here:
http://docs.continuum.io/anaconda/install

You may need to update setup_centos5.sh to change the directory path if 
you want to install anaconda python to a different directory.

If you are interested in devsim, and not using symdiff by itself, you 
can get a binary release from:

https://github.com/devsim/devsim/releases/latest

You would then set your path to anaconda python using:
#bash
source  devsim/anaconda_vars.sh
#csh
source devsim/anaconda_vars.csh

If you have anaconda installed in a different place, adjust these 
scripts as necessary.

chihiro0019, 2015-11-10:
Thank you for you advice.

I installed Anaconda2-2.4.0-Linux-x86_64.sh as shown below
(I tried Anaconda3-2.4.0-Linux-x86_64.sh, too, the same result )

>
ls /root
Desktop              anaconda-ks.cfg  anaconda3_3.5  install.log.syslog  
RPM-GPG-KEY.dag.txt  anaconda2        install.log    mbox

I ran scripts/setup_centos_5.sh again.

>
sh scripts/setup_centos_5.sh 
...
CMake Warning (dev) in CMakeLists.txt:
  No cmake_minimum_required command is present.  A line of code such as
    cmake_minimum_required(VERSION 2.8)
  should be added at the top of the file.  The version specified may be lower
-- Configuring done
-- Generating done
CMake Warning:
  Manually-specified variables were not used by the project:
    PYTHON_BIN
-- Build files have been written to: /install/devsim-master/external/symdiff/linux_x86_64_debug
...
-- Configuring done
-- Generating done
CMake Warning:
  Manually-specified variables were not used by the project:
    PYTHON_BIN
-- Build files have been written to: /install/devsim-master/external/symdiff/linux_x86_64_release

Then, there is no improvement in making symdiff

>
make -j2
[ 46%] Built target symdiff
[ 93%] Built target symdiff_static
make[2]: *** `src/tclcomp/libsymdifftcl.so'  no rule to make  `/root/anaconda/lib/libtclstub8.5.a' 
[ 94%] make[2]: *** `src/pycomp/symdiff.so'  no rule to make  `/root/anaconda/lib/libpython2.7.so' 
[ 96%] Building CXX object src/tclcomp/CMakeFiles/symdiff_tcl.dir/tclcomp.o
Building CXX object src/pycomp/CMakeFiles/symdiff_py.dir/PythonSymdiffCommands.o
/install/devsim-master/external/symdiff/src/pycomp/PythonSymdiffCommands.cc:23:20: error: 
...

juanesanchez, 2015-11-10:
The installation assumes that Anaconda is installed in your home directory:
/root/anaconda

You need to verify these files exist:
/root/anaconda/lib/libtclstub8.5.a
/root/anaconda/lib/libpython2.7.so

If they are installed somewhere else, modify:
setup_centos_5.sh

You may have to delete the linux_x86_64_release and start again or modify linux_x86_64_release/CMakeCache.txt

and modify the entries for:
TCL_ARCHIVE
TCL_INCLUDE
TCL_BIN
PYTHON_ARCHIVE
PYTHON_INCLUDE
PYTHON_BIN

chihiro0019, 2015-11-10:
Thank you, I could compile symdiff, but failed to ctest of symdiff
and making devsim failed at 99%. I attached 3 logs below.
Please advice.
                                  Kimihiro

symdiff make
>
pwd
/install/devsim-master/external/symdiff/linux_x86_64_release
make -j2
[ 46%] Built target symdiff
[ 93%] Built target symdiff_static
[ 94%] [ 96%] Building CXX object src/pycomp/CMakeFiles/symdiff_py.dir/PythonSymdiffCommands.o
Building CXX object src/tclcomp/CMakeFiles/symdiff_tcl.dir/tclcomp.o
Linking CXX shared library libsymdifftcl.so
[100%] Built target symdiff_tcl
Linking CXX shared library symdiff.so
[100%] Built target symdiff_py

symdiff test
>
pwd
/install/devsim-master/external/symdiff/linux_x86_64_release
ctest -j2
Start processing tests
Test project /install/devsim-master/external/symdiff/linux_x86_64_release
  1/ 19 Testing parsetest_tcl                 ***Failed 
  2/ 19 Testing symdiff1                      ***Failed 
  3/ 19 Testing symdiff2                      ***Failed 
  4/ 19 Testing symdiff3                      ***Failed 
  5/ 19 Testing symdiff4                      ***Failed 
  6/ 19 Testing arrhenius_tcl                 ***Failed 
  7/ 19 Testing units                         ***Failed 
  8/ 19 Testing table                         ***Failed 
  9/ 19 Testing parsetest_py                  ***Failed 
 10/ 19 Testing deftest_py                    ***Failed 
 11/ 19 Testing arrhenius_py                  ***Failed 
 12/ 19 Testing modellist1_py                 ***Failed 
 13/ 19 Testing ordered_py                    ***Failed 
 14/ 19 Testing remove1_py                    ***Failed 
 15/ 19 Testing subexpression1_py             ***Failed 
 16/ 19 Testing undefined1_py                 ***Failed 
 17/ 19 Testing utf8_py                       ***Failed 
 18/ 19 Testing models1_py                    ***Failed 
 19/ 19 Testing table_py                      ***Failed 

devsim make
>
pwd
/install/devsim-master/linux_x86_64_release
make -j2
...
[ 99%] Built target commands
Scanning dependencies of target devsim_py
Scanning dependencies of target devsim_tcl
make[2]: *** `src/main/devsim_tcl'  no rule to make target  `/opt/intel/compilers_and_libraries/linux/mkl/lib/intel64/libmkl_intel_lp64.a' 
[ 99%] make[2]: *** `src/main/devsim_py'  no rule to make target `/opt/intel/compilers_and_libraries/linux/mkl/lib/intel64/libmkl_intel_lp64.a' 
[100%] Building CXX object src/main/CMakeFiles/devsim_tcl.dir/devsim_tcl.o
Building CXX object src/main/CMakeFiles/devsim_py.dir/devsim_py.o

juanesanchez, 2015-11-10:
What do you see when you run:
symdiff/bin/symdiff

The centos 5 version assumes that you have installed the Intel Math 
Kernel library on your build system.  It can be downloaded for free 
under the community edition from:
https://software.intel.com/en-us/articles/free_mkl

If you want to choose different math libraries.  Create a file named 
local.cmake (or edit cmake/centos5.cmake) and set:
SET (BLAS_ARCHIVE /path/to/blas /path/to/lapack)
or any other version of lapack and blas that you prefer.

For example:
SET (BLAS_ARCHIVE -lblas -llapack)

chihiro0019, 2015-11-11:
Thank you for your advice, I could do symdiff ctest.
I could make devsim, but failed to run it.

>
pwd
/home/install/devsim-master/linux_x86_64_release/src/main
./devsim_py
./devsim_py: error while loading shared libraries: libpython2.7.so.1.0: cannot open shared object file: No such file or directory

But, before this
>
yum install python2.7 python2.7-devel 
Loaded plugins: fastestmirror
Loading mirror speeds from cached hostfile
 * base: ftp.riken.jp
 * epel: ftp.riken.jp
 * extras: ftp.riken.jp
 * updates: ftp.riken.jp
Excluding Packages in global exclude list
Finished
Setting up Install Process
No package python2.7 available.
No package python2.7-devel available.
Nothing to do

>
yum install python2.7-libs
Loaded plugins: fastestmirror
Loading mirror speeds from cached hostfile
 * base: ftp.riken.jp
 * epel: ftp.kddilabs.jp
 * extras: ftp.riken.jp
 * updates: ftp.riken.jp
Excluding Packages in global exclude list
Finished
Setting up Install Process
No package python2.7-libs available.
Nothing to do

juanesanchez, 2015-11-11:
Anaconda should provide the python 2.7 libraries in ${HOME}/anaconda.  
You can then source:
for csh:
source scripts/anaconda_vars.csh
for bash:
source scripts/anaconda_vars.sh

If the path is incorrect you can adjust it in the scripts accordingly.  
You can try:
ldd devsim_py

chihiro0019, 2015-11-11:
Thank you for the quick response.

I have anaconda directory as

>
pwd
/root
ls -ld anaconda*
lrwxrwxrwx  1 root root    9 11月 10 13:43 anaconda -> anaconda2
-rw-------  1 root root 1295  3月 22  2013 anaconda-ks.cfg
drwxr-xr-x 16 root root 4096 11月 10 09:45 anaconda2
drwxr-xr-x 16 root root 4096 11月 10 09:20 anaconda3

But, when I run devsim_py

>
devsim_py
----------------------------------------
 DEVSIM
 Version: Beta 0.01
 Copyright 2009-2015 Devsim LLC
----------------------------------------
Python 2.7.10 |Anaconda 2.4.0 (64-bit)| (default, Oct 19 2015, 18:04:42) 
[GCC 4.4.7 20120313 (Red Hat 4.4.7-1)] on linux2
Type "help", "copyright", "credits" or "license" for more information.
Anaconda is brought to you by Continuum Analytics.
Please check out: http://continuum.io/thanks and https://anaconda.org

Do I need set up something ? ctest -j2 all failes

>
pwd
/home/install/devsim-master/linux_x86_64_release
ctest -j2
Start processing tests
Test project /install/devsim-master/linux_x86_64_release
  1/ 92 Testing testing/cap2                  ***Failed 
  2/ 92 Testing testing/equation1             ***Failed 
  3/ 92 Testing testing/ptest1                ***Failed 
  4/ 92 Testing testing/ptest2                ***Failed 
  5/ 92 Testing testing/sqlite2               ***Failed 
  6/ 92 Testing testing/sqlite3               ***Failed 
  7/ 92 Testing testing/testfunc              ***Failed 
  8/ 92 Testing testing/utf8_2                ***Failed 
  9/ 92 Testing testing/laux1                 ***Failed

juanesanchez, 2015-11-11:
Look at the log files in:
linux_x86_64_release/Testing/Temporary/

and see what the errors are.

chihiro0019, 2015-11-12:
DevSim ctest -j2 failed, but there is no error message in

>pwd
/home/install/devsim-master/linux_x86_64_release/Testing/Temporary
[root@gstj01 Temporary]# ls -l
total 96
drwxr-xr-x 2 root root  4096 11月 12 10:15 .
drwxr-xr-x 3 root root  4096 11月 11 10:44 ..
-rw-r--r-- 1 root root 81212 11月 12 10:15 LastTest.log
-rw-r--r-- 1 root root  2160 11月 12 10:15 LastTestsFailed.log

>
grep -i error < LastTest.log
grep -i error < LastTestsFailed.log

>
which devsim_py
/install/devsim-master/linux_x86_64_release/src/main/devsim_py

juanesanchez, 2015-11-12:
try:
more LastTest.log
more LastTestsFailed.log

or open the files in a text editor.

chihiro0019, 2015-11-12:
Yes, I found 'failed' in LastTest.log which says line 18 of 
rundifftest.pl failed as

/home/install/devsim-master/linux_x86_64_release/Testing/Temporary/LastTest.log
>
/install/devsim-master/linux_x86_64_release/src/main/devsim_py cap2.py failed to run at /install/devsim-master/testing/rundifftest.pl line 18.

/home/install/devsim-master/testing/rundifftest.pl
>
line18: my $retval = (system($command)==0) or die "$testexe failed to run";

juanesanchez, 2015-11-12:
please try:
cd devsim-master/testing
/install/devsim-master/linux_x86_64_release/src/main/devsim_py cap2.py

or:
../linux_x86_64_release/src/main/devsim_py cap2.py

chihiro0019, 2015-11-12:
I'm not sure, but some setting is wrong.

>
cd devsim-master/testing
/install/devsim-master/linux_x86_64_release/src/main/devsim_py cap2.py
/install/devsim-master/linux_x86_64_release/src/main/devsim_py: error while loading shared libraries: libpython2.7.so.1.0: cannot open shared object file: No such file or directory

>
pwd
/root/anaconda
ls -ld lib/libpy*
lrwxrwxrwx 1 root root      19 11月 10 09:45 lib/libpython2.7.so -> libpython2.7.so.1.0
-rwxrwxr-x 2 root root 6545143 10月 20 08:07 lib/libpython2.7.so.1.0

juanesanchez, 2015-11-13:
Try:
ldd /install/devsim-master/linux_x86_64_release/src/main/devsim_py

to see all the missing libraries.

You can:
source devsim-master/scripts/anaconda_vars.sh

to set the appropriates paths.  You should see no errors then with ldd.

chihiro0019, 2015-11-13:
Yes. There is no ldd error. But failed to ctest -j2.

>
pwd
/home/install
source devsim-master/scripts/anaconda_vars.csh

>
ldd /install/devsim-master/linux_x86_64_release/src/main/devsim_py 
        linux-vdso.so.1 =>  (0x00007fffcc7fd000)
        libpython2.7.so.1.0 => /root/anaconda/lib/libpython2.7.so.1.0 (0x00002aee85fb9000)
        libpthread.so.0 => /lib64/libpthread.so.0 (0x0000003aaf200000)
        libstdc++.so.6 => /usr/lib64/libstdc++.so.6 (0x0000003ab4a00000)
        libm.so.6 => /lib64/libm.so.6 (0x0000003aae600000)
        libgcc_s.so.1 => /lib64/libgcc_s.so.1 (0x0000003ab4200000)
        libc.so.6 => /lib64/libc.so.6 (0x0000003aae200000)
        /lib64/ld-linux-x86-64.so.2 (0x0000003aade00000)
        libdl.so.2 => /lib64/libdl.so.2 (0x0000003aaea00000)
        libutil.so.1 => /lib64/libutil.so.1 (0x0000003abd400000)

>
cd devsim-master/testing 
/install/devsim-master/linux_x86_64_release/src/main/devsim_py cap2.py
 ...

devsim_py cap2.py looks working, but, ctest doesn't

>
 pwd
/home/install/devsim-master/linux_x86_64_release
[root@gstj01 linux_x86_64_release]# ctest -j2
Start processing tests
Test project /install/devsim-master/linux_x86_64_release
  1/ 92 Testing testing/cap2                  ***Failed 
  2/ 92 Testing testing/equation1             ***Failed 
  3/ 92 Testing testing/ptest1                ***Failed 

>
which devsim_py
/install/devsim-master/linux_x86_64_release/src/main/devsim_py
[root@gstj01 linux_x86_64_release]# devsim_py
 ----------------------------------------
 DEVSIM
 Version: Beta 0.01
 Copyright 2009-2015 Devsim LLC
 ----------------------------------------
Python 2.7.10 |Anaconda 2.4.0 (64-bit)| (default, Oct 19 2015, 18:04:42) 
[GCC 4.4.7 20120313 (Red Hat 4.4.7-1)] on linux2
Type "help", "copyright", "credits" or "license" for more information.
Anaconda is brought to you by Continuum Analytics.
Please check out: http://continuum.io/thanks and https://anaconda.org
>>>

juanesanchez, 2015-11-13:
It sounds like the problem is compiled correctly and is able to run.

Please look at the test logs to see if the messages are different.  I 
suspect that the test is running, but the golden test results need to be 
downloaded.

Please download the centos test results from here and follow the 
instructions on the main page (README.txt).
https://github.com/devsim/devsim_tests_centos5

essentially:
mv devsim_tests_centos5 devsim-master/goldenresults/centos_5_x86_64

Most of the tests should pass, except for the 3 listed in README.txt.

Note that these tests rely on an exact numerical match, which varies on 
the processor and math libraries used to compile the program.  These 
test results where generated for centos 5 running in a vmware image on 
an Intel i7 Macbook Pro.

chihiro0019, 2015-11-13:
Thank you, now it seems running.
By the way, I installed devsim as the root with home directory /root.
If I want to run devsim as an usual user, maybe I need to change $HOME
to /root in many scripts, do I ?
Could you advice which files I need to change ?

juanesanchez, 2015-11-13:
You can rerun the anaconda installer and install it somewhere like:
/opt/anaconda
so that all users have access to it.

Then create a package:
cd devsim/dist
bash ./package_linux.sh devsim

It then creates a tgz file you can uncompress wherever you like:
devsim_x86_64_20151113.tgz
or:
mv ./devsim_x86_64 /opt

The anaconda_vars.csh script will then be in the base directory of the 
installation.

Modify the anaconda_vars script to point to the new anaconda location.

For example:
setenv LD_LIBRARY_PATH /opt/anaconda/lib
setenv PYTHONHOME /opt/anaconda
setenv TCL_LIBRARY /opt/anaconda/lib/tcl8.5

chihiro0019, 2015-11-13:
Thank you. Now it's working in an usual user's environment.
I noticed 2 things about 
/home/install/devsim-master/devsim/dist/package_linux.sh

1. SRC_DIR=../${PLATFORM}_${ARCH}_release/src/main

must be ../../${PLATFORM}_${ARCH}_release/src/main

2 . The name of binary was not devsim_py
>
pwd
/home/install/devsim_x86_64/bin
[root@gstj01 bin]# ls
.  ..  devsim  devsim_tcl

juanesanchez, 2015-11-13:
I am glad you got it working.  I'll look into improving the packaging script.  It was intentional that devsim_py is renamed devsim since that is the preferred version of the software.  devsim_tcl is a legacy program to run the complete suite of tests.

chihiro0019, 2015-11-14:
Thank you very much, I close this issue.

One problem I met in my Paraview binary installation is below.
I needed to change file permission.

>
pwd
/home/install/ParaView-4.4.0-Qt4-Linux-64bit/lib/paraview-4.4
ls -l libG*
-rwx------ 1 1001 1001 378310  Sep 14 22:36 libGenericIO.so

closed 2015-11-14 by chihiro0019

juanesanchez, 2015-11-14:
I prefer using Visit to Paraview.  I have had the best luck using 
Tecplot output instead of Vtk.

If you set group permissions, or all permissions for Paraview.
chmod -R go+rx /home/install/ParaView-4.4.0-Qt4-Linux-64bit

You can get Visit from here:
https://wci.llnl.gov/simulation/computer-codes/visit/