devsim/devsim #16: Which option should be chosen to plot ***.dat ?

chihiro0019, 2016-01-05:
I simulated and tried to plot /devsim_x86_64/examples/diode/gmsh_diode3d_dd.dat, but Paraview asks me to choose one of (ICM Files, Overflow Files, Omicron Model Input, ParaDis Files,,, so on). Which one should I choose? Or, do I need to install Tecplot?

juanesanchez, 2016-01-05:
I apologize for not having a consistent naming extension for the different
file formats.  It is in the Tecplot format, as can be seen from this line
in the input file:
write_devices(file="gmsh_diode3d_dd.dat", type="tecplot")

You should be able to open it in Paraview with the Tecplot format.  You can
also try using VisIt from here:
https://*visit*.*llnl*.gov/

I have had the best experience with the VisIt viewer with the Tecplot file
format.

chihiro0019, 2016-01-06:
When I tried to use paraview/tecplot, I met many errors for diodes as

>
ERROR: In /home/utkarsh/public-cmbsuperbuild/build/paraview/src/paraview/VTK/IO/Geometry/vtkTecplotReader.cxx, line 1913
vtkTecplotReader (0x146e1780): /install/devsim_x86_64/examples/diode/gmsh_diode3d_dd.dat: The record type 1 found in the file is unknown.

Do I need to get tecplot ? or any other tool ?

juanesanchez, 2016-01-06:
please try VisIt instead of Paraview

chihiro0019, 2016-01-06:
I downloaded visit binary files from https://wci.llnl.gov/simulation/computer-codes/visit/downloads.
But I met many errors when I ran Visit    /home/install/visit2_10_0.linux-x86_64/bin/visit

for example,
>
/home/install/visit2_10_0.linux-x86_64/2.10.0/linux-x86_64/bin/gui: /lib64/libc.so.6: version `GLIBC_2.9' not found (required by /home/install/visit2_10_0.linux-x86_64/2.10.0/linux-x86_64/lib/libQtGui.so.4)

>
ls /lib64/libc.so*
/lib64/libc.so.6

Do you know anything about this ?

juanesanchez, 2016-01-06:
Unfortunately it looks like those symbols would only be available in a 
newer Linux operating system.

It looks like visit 2.6.3 is supported for Redhat 5, if that is what you 
are running.  It is found lower on the download page.

Please note that Visit would also work on the Windows operating system.

Another alternative is to use Paraview with the vtk format.  It may work 
work for you.

Please try this line in your input file:
write_devices(file="gmsh_diode3d_dd", type="vtk")

chihiro0019, 2016-01-06:
Yes, after my question, I have already tried Paraview as you mentioned and it did work. Thanks.

closed 2016-01-06 by chihiro0019