devsim/devsim #17: DevSim Transient example

chihiro0019, 2016-01-07:
I want to run transient simulation with DevSim.
Don't you have any example ?
I tried to modify gmsh_diode3d.py, but I don't know how to do it.

How can I define transient voltage source, start time, end time and intervals ?

By the way, does DevSim use [MKS] unit systems ?

>
solve(type="transient_dc", absolute_error=1.0, relative_error=1e-12, maximum_iterations=30)
solve(type="transient_tr", absolute_error=1e10, relative_error=1e-8, maximum_iterations=50)
v = 0.1
while v < 0.51:
  set_parameter(device=device, name=GetContactBiasName("top"), value=v)
  solve(type="dc", absolute_error=1e10, relative_error=1e-8, maximum_iterations=30)
  PrintCurrents(device, "top")
  PrintCurrents(device, "bot")
  v += 0.1

juanesanchez, 2016-01-07:
In the devsim/testing directory, there are a few examples:
transient_circ.tcl
transient_circ2.tcl
transient_circ3.tcl

They are simple circuit examples using the old Tcl interface to devsim.  
They should be straightforward to convert to Python.

To have a changing voltage source, you would need to write your own loop 
to track time.

I need to create additional transient examples, using a loop, in Python.

For most of the examples, I assume that the physical parameters and 
lengths are in terms of cm.  For example
ni = 1e10 /cm^3
mu_n = 1000 cm^2/(V s)

chihiro0019, 2016-01-07:
Thank you. I took a look into one of testing tcl, but it was too simple, so I am waiting for your new example.

chihiro0019, 2016-01-12:
I got an e-mail from you, just now, but I can't understand what you'd like to tell me. Is anything new ? about transient example ?

>
Documentation:
devsim.pdf
Additional Information:
http://www.devsim.org
Forum:
https://groups.google.com/forum/#!forum/devsim
Supported Platforms:
...

juanesanchez, 2016-01-12:
Hi,

This email was automatically generated because you are watching the 
project, and a new release was created.

I have not created the transient example yet.  I hope to find some time 
soon to put an example together.

The main change to the code is that the help documentation is now 
integrated into the python interpreter.

The windows and linux binary release use the newest version of the Intel 
Math Kernel Library.  The windows version was built using the newest 
Microsoft Visual Studio compiler.

chihiro0019, 2016-01-12:
I see, thanks. Waiting for your update.

chihiro0019, 2016-01-13:
Sorry to bother you, but except transient, I have several questions about mesh I'd like to ask you.

On figure 3.1 in DevSim manual, I added n, p, Ψ and E, are they correct?

[image]

In your cap2d example, there are x-only and y-only mesh definitions independently each other.
>
create_2d_mesh(mesh=device)
add_2d_mesh_line(mesh=device, dir="y", pos=ymin, ps=0.1)
add_2d_mesh_line(mesh=device, dir="y", pos=y1  , ps=0.1)
add_2d_mesh_line(mesh=device, dir="y", pos=y2  , ps=0.1)
add_2d_mesh_line(mesh=device, dir="y", pos=y3  , ps=0.1)
add_2d_mesh_line(mesh=device, dir="y", pos=y4  , ps=0.1)
add_2d_mesh_line(mesh=device, dir="y", pos=ymax, ps=5.0)
>
device=device
region="air"
>
add_2d_mesh_line(mesh=device, dir="x", pos=xmin, ps=5)
add_2d_mesh_line(mesh=device, dir="x", pos=x1  , ps=2)
add_2d_mesh_line(mesh=device, dir="x", pos=x2  , ps=0.05)
add_2d_mesh_line(mesh=device, dir="x", pos=x3  , ps=0.05)
add_2d_mesh_line(mesh=device, dir="x", pos=x4  , ps=2)
add_2d_mesh_line(mesh=device, dir="x", pos=xmax, ps=5)

Does DevSim generates rectangular or triangle mesh?  The plot by Paraview seems to show it is triangles.
Then, how can DevSim generate these triangles ?
And if triangles, what does node volume look like? 

[image]

[image]

In your gmsh_diode3d example, I can see complicated triangle mesh not from rectangular mesh.
What is the policy of generating this mesh? And how? This may be impossible with DevSim functions.

[image]

[image]

juanesanchez, 2016-01-14:
Hello,

For your questions, please see my response here:
https://groups.google.com/forum/?hl=en#!category-topic/devsim/all-questions/aFbhkIVs4IM

Please join the DEVSIM google group, and feel free to ask more questions.

chihiro0019, 2016-01-15:
Thank you for your reply. I joined DEVSIM google group, but since I'm not used to be in this group, let me write my requests here again.
1 Please make some transient examples.
2 Please give me .geo file for gmsh_diode3d

I always appreciate for your kind and quick support.

juanesanchez, 2016-01-15:
I have created Issue #18 for your 2nd request.  Please subscribe for notifications

chihiro0019, 2016-01-19:
How about transient example? if possible i'd like to get it this week.
I will upgrade CentOS next month.

juanesanchez, 2016-01-20:
I am going to need some more time.  I won't be able to get it done this 
week.

chihiro0019, 2016-01-27:
By the way, don't you recommend me some good book to understand VTK and Gmsh?

juanesanchez, 2016-01-28:
I am still working on developing a transient example.  I hope to be able 
to get you something within the next 2 weeks.

Please ask non-bug related questions on the mailing list here:
https://groups.google.com/forum/#!forum/devsim

chihiro0019, 2016-02-08:
How about your progress? I'd like to try transient at least 2-3 days before Feb.25.

closed 2017-02-27 by tcaduser