devsim/devsim #18: 3D Diode Example

juanesanchez, 2016-01-15:
@chihiro0019 requests the gmsh file used to generate the 3D diode example.

chihiro0019, 2016-01-15:
Yes, thank you for your action. What should I do ?

juanesanchez, 2016-01-16:
I think I have a file that will generate a structure similar to the 3d diode.  I will try to get that to you soon.

juanesanchez, 2016-01-17:
I have uploaded a gmsh file which creates a mesh for block with a top and bottom contact.  It is similar to the one I used to generate the original 3d diode example.
https://github.com/devsim/devsim/blob/master/examples/diode/3dblock.geo

chihiro0019, 2016-01-17:
Thank you, I will analyze this.

chihiro0019, 2016-01-18:
I analyzed your 3d-diode.geo example, but it doesn't use tetrahedron but cuboid only.

[image]

[image]

juanesanchez, 2016-01-18:
How did you generate the figure with the surface loops? I never saw that
feature before.

Gmsh has several options.  It should be possible to set the meshing options
so that tetrahedra are used for the volume mesh.  Is the diode result you
show from the new geo file?  It should have tetrahedra in order to work

chihiro0019, 2016-01-18:
I see. Thanks.
Unfortunately, installed Gmsh didn't run by some library errors on our CentOS 5.1.
I don't know anything about Gmsh function. The figure I show is what I drew reading
.geo input file.

juanesanchez, 2016-01-19:
centos 5 is a very old system. You may be able to install the gmsh package
for your system, or use gmsh on another Windows or Linux machine

chihiro0019, 2016-02-01:
Sorry for my novice question, but I am a little confused with mesh formats, 1 DevSim 2 Genius 3 Gmsh.

In DevSim cap2d.py, this is described
>
write_devices(file="cap2d.msh", type="devsim")

and the generated cap2d.msh file contains
>
begin_device "MyDevice"
begin_coordinates
-2.500000000000000e+01	0.000000000000000e+00	0.000000000000000e+00
-2.500000000000000e+01	1.000000000000000e-01	0.000000000000000e+00
...

On the other hand, in gmsh_diode3d.py
>
create_gmsh_mesh (mesh="diode3d", file="gmsh_diode3d.msh")

and the generated gmsh_diode3d.msh contains
>
$MeshFormat
2.1 0 8
$EndMeshFormat
$PhysicalNames

and this format looks as explained in Gmsh 2.11 manual.

I have another example (not from you) which contains
>
begin_device "mos"
begin_coordinates
0.000000000000000e+00	2.500000000000000e-01	-3.500000000000000e-01
6.550019000000000e-02	2.500000000000000e-01	-3.500000000000000e-01

and this looks as the same as cap2d.msh format, but however in 3D. Is this DevSim mesh format? If so, where can I get the manual to understand this format?

chihiro0019, 2016-02-03:
Related to my previous question, DevSim mesh format, I have other questions here.

Q1: For the case of cap2d, in cap2d.py written the descriptions below.

ex-a
>
xmin=-25, x1  =-24.975, x2  =-2, x3  =2, x4  =24.975, xmax=25.0
ymin=0.0, y1  =0.1, y2  =0.2, y3  =0.8, y4  =0.9, ymax=50.0
create_2d_mesh(mesh=device)
add_2d_mesh_line(mesh=device, dir="y", pos=ymin, ps=0.1)
add_2d_mesh_line(mesh=device, dir="y", pos=y1  , ps=0.1)
add_2d_mesh_line(mesh=device, dir="y", pos=y2  , ps=0.1)
add_2d_mesh_line(mesh=device, dir="y", pos=y3  , ps=0.1)
add_2d_mesh_line(mesh=device, dir="y", pos=y4  , ps=0.1)
add_2d_mesh_line(mesh=device, dir="y", pos=ymax, ps=5.0)

But, at the same time, generated cap2d.msh has descriptions as

ex-b
>
begin_coordinates
-2.500000000000000e+01	0.000000000000000e+00	0.000000000000000e+00
-2.500000000000000e+01	1.000000000000000e-01	0.000000000000000e+00
-2.500000000000000e+01	2.000000000000000e-01	0.000000000000000e+00
-2.500000000000000e+01	3.000000000000000e-01	0.000000000000000e+00
-2.500000000000000e+01	4.000000000000000e-01	0.000000000000000e+00
-2.500000000000000e+01	5.000000000000000e-01	0.000000000000000e+00
-2.500000000000000e+01	6.000000000000000e-01	0.000000000000000e+00
-2.500000000000000e+01	7.000000000000000e-01	0.000000000000000e+00
-2.500000000000000e+01	8.000000000000000e-01	0.000000000000000e+00
-2.500000000000000e+01	9.000000000000000e-01	0.000000000000000e+00
-2.500000000000000e+01	1.000000000000000e+00	0.000000000000000e+00
-2.500000000000000e+01	1.111111111111111e+00	0.000000000000000e+00
-2.500000000000000e+01	1.234567901234568e+00	0.000000000000000e+00
-2.500000000000000e+01	1.371742112482853e+00	0.000000000000000e+00
-2.500000000000000e+01	1.524157902758726e+00	0.000000000000000e+00

In ex-b from y=1.00 to lower, it doesn't seem to have step size (ps) 0.1, why?
Where is the command which create those intervals?

Q2: It seems in ex-b, from top to bottom, it has sequential node number from 1 to N as
shown in ex-c. Is this correct ?

ex-c
>
node x y z
0 -2.500000000000000e+01	0.000000000000000e+00	0.000000000000000e+00
1 -2.500000000000000e+01	1.000000000000000e-01	0.000000000000000e+00
2 -2.500000000000000e+01	2.000000000000000e-01	0.000000000000000e+00

Q3: If Q2=yes, then ex-d seems to list up nodes (defined in ex-c) in "air"

ex-d
>
begin_region "air" "gas"
begin_nodes
0
1
49
50

Q4: if Q2=yes then ex-e can be interpreted as e-f. Is this correct?

ex-e
>
begin_edges
0	1
0	2
1	2
1	3
1	4

ex-f
>
edge node1 node2
0  0	1
1  0	2
2  1	2
3  1	3
4  1	4

Q5: if Q2=yes then ex-g can be interpreted as e-h. Is this correct? And numbers are
for node or edge? I guess they are edges.

ex-g
>
begin_triangles
0	1	2
1	2	3
1	3	4
2	3	98
3	4	5
3	98	99

ex-h
>
triangle node1 node2 node3 (?)
triangle edge1 edge2 dege3 (?)
0  0	1	2
1  1	2	3
2  1	3	4
3  2	3	98
4  3	4	5
5  3	98	99

juanesanchez, 2016-02-03:
Hello,

Sorry for the delay in the response.  Please see my responses inline.

> Related to my previous question, DevSim mesh format, I have other
> questions here.
>
> Q1: For the case of cap2d, in cap2d.py written the descriptions below.
>
> ex-a
>
>     xmin=-25, x1 =-24.975, x2 =-2, [...]
At
y=0.9,  ps=0.1
y=50, ps=5.0

That means that the spacing is increasing from 0.1 at 0.9 to 5.0  at 
50.  The spacing is increased between those two lines

Please see the functions here for the implementation:
https://github.com/devsim/devsim/blob/master/src/meshing/MeshUtil.cc#L80
https://github.com/devsim/devsim/blob/master/src/meshing/MeshUtil.cc#L49

> Q2: It seems in ex-b, from top to bottom, it has sequential node
> number from 1 to N as
> shown in ex-c. Is this correct ?
>
> ex-c
>
>     node x y z
>     0 -2.500000000000000e+01 0.000000000000000e+00 [...]

These are coordinates.  This is the set of all the unique points in the 
entire mesh.  They are N coordinates numbered from 0 to N-1

> Q3: If Q2=yes, then ex-d seems to list up nodes (defined in ex-c) in "air"
>
> ex-d
>
>     begin_region "air" "gas"
>     begin_nodes
>     0
>     1
>     49
>     50
The nodes are numbered 0 to N-1.  Each node index refers to a coordinate.

> Q4: if Q2=yes then ex-e can be interpreted as e-f. Is this correct?
>
> ex-e
>
>     begin_edges
>     0 1
>     0 2
>     1 2
>     1 3
>     1 4

> ex-f
>
>     edge node1 node2
>     0 0 1
>     1 0 2
>     2 1 2
>     3 1 3
>     4 1 4

This is correct.  The edge refers to pairs of nodes in the region.

> Q5: if Q2=yes then ex-g can be interpreted as e-h. Is this correct?
> And numbers are
> for node or edge? I guess they are edges.
>
> ex-g
>
>     begin_triangles
>     0 1 2
>     1 2 3
>     1 3 4
>     2 3 98 [...]
A triangle refers to nodes.

The interface and contact sections of the file format also refers to nodes.

Please let me know if you have any additional questions.

chihiro0019, 2016-02-03:
Thank you for your explanation and I guess there is no manual for this format, isn't it?
>
That means that the spacing is increasing from 0.1 at 0.9 to 5.0  at
50.  The spacing is increased between those two lines

Does this mean points are generated by equal rate from 0.1 to 50?

juanesanchez, 2016-02-04:
On 2/3/16 12:57 AM, Kimihiro wrote:
>
> Thank you for your explanation and I guess there is no manual for this 
> format, isn't it?
>
>     That means that the spacing is increasing from 0.1 at 0.9 to 5.0 at
>     50. The spacing is increased between those two lines
>
> Does this mean points are generated by equal rate from 0.1 to 50?
>
It is a geometric progression.  The spacing varies as:
dx(i) = sl*pow(r, i)

where i is an integer from 0, N-1, and r is chosen such that:
dx(0) = sl
dx(N-1) = sh

where sl and sh are the initial and final spacing between the 2 lines.

x(0) = xl
x(1) = xl + dx(0)
x(2) = x(1) + dx(1)
x(i) = x(i) + dx(i)
x(N-1) = xh

where xl and xh are the 2 line positions  It can be shown that

L = xh - xl
r = (L - sl)/(L - sh)
N=floor(log(sh/sl)/log(r)))+1

chihiro0019, 2016-02-04:
Thank you, It made it clearer.

chihiro0019, 2016-02-04:
May I ask one more?
In the example below, is each line data of Acceptors and Donors 
at each coordinate, 0, 1, 2 ... sequentially?

>
begin_device "mos"
begin_coordinates
0.000000000000000e+00	2.500000000000000e-01	-3.500000000000000e-01
6.550019000000000e-02	2.500000000000000e-01	-3.500000000000000e-01
6.550019000000000e-02	2.577895000000000e-01	-3.500000000000000e-01
…
…
begin_node_model "Acceptors"
DATA
1.993858e+17
2.001914e+17
1.436887e+17
1.428246e+17
2.001752e+17
1.437125e+17
2.001061e+17
…
…
begin_node_model "Donors"
DATA
1.216772e+05
1.210303e+05
1.512096e+05
1.515655e+05
1.210448e+05
1.511929e+05
…

juanesanchez, 2016-02-04:
The node data corresponds to the node entries.  The node entries have a 
corresponding coordinate.

For the first data point you would look up the first node.  The first 
node has an index into the list of all the coordinates.

Are you trying to import or export data from DEVSIM?  It is easy to 
export using a Python script.

chihiro0019, 2016-02-04:
Thank you, I got it.
For this specific case, This is 3D MOS and I only have *.msh to import.
I'm not sure how this .msh was created, since DevSim has 2D meshing only.

juanesanchez, 2016-02-05:
Please look at:
devsim/testing/gmsh_mos3d.geo

The Tcl (not Python) version to take the file from a gmsh msh to a 
devsim msh is:
gmsh_mos3d_create.tcl

And a basic simulation is:
gmsh_mos3d.tcl

closed 2017-02-27 by tcaduser