devsim/devsim #19: About mesh format

chihiro0019, 2016-02-24:
Sorry, but I need to know right now.

A mesh file has descriptions below. In this case, for example 7116 in 
'begin_contact "drain" "Si" "metal"' is the 7116'th node in 'begin_coordinates'
or the 7116'th node in 'begin_region "Si" "Silicon"' ?
Please advice !!

>
begin_device "mos"
begin_coordinates
0.000000000000000e+00	2.500000000000000e-01	-3.500000000000000e-01
6.550019000000000e-02	2.500000000000000e-01	-3.500000000000000e-01
...
begin_region "Si" "Silicon"
begin_nodes
0
1
...
24554
end_nodes
...
begin_contact "drain" "Si" "metal"
begin_nodes
6967
6968
...
7116
end_nodes
...

juanesanchez, 2016-02-24:
A contact only connects to 1 region.

In this case Contact "drain" references nodes in region "Si".

For Contact node 7116.

Look at the 7117th position in region "Si".  This is since it is numbered:
0
1
2
...
7116

You may be able to do this much easier from the python interface.

chihiro0019, 2016-02-24:
Thank you very much. I have another quick question.
Does this description refers to coordinates?
There is no region specification here.

>
begin_node_model "Donors"
DATA
1.216772e+05
1.210303e+05
1.512096e+05
1.515655e+05
1.210448e+05

juanesanchez, 2016-02-24:
begin_node_model is located in a region block:
begin_region
...
begin_node_model
...
end_node_model
...
end_region

juanesanchez, 2016-02-24:
Here is a python script that will print the node index, coordinate index, and the position for a contact:
```
from ds import *
load_devices(file="mos_2d_dd.msh")

device="mymos"

contact_node_model(device=device, contact="drain", name="at_drain", equation="1")
contact_node_model(device=device, contact="drain", name="drain_ni", equation="node_index")
contact_node_model(device=device, contact="drain", name="drain_ci", equation="coordinate_index")

at_drain = get_node_model_values(device=device, region="bulk", name="at_drain")
drain_ni = get_node_model_values(device=device, region="bulk", name="drain_ni")
drain_ci = get_node_model_values(device=device, region="bulk", name="drain_ci")
x = get_node_model_values(device=device, region="bulk", name="x")
y = get_node_model_values(device=device, region="bulk", name="y")
z = get_node_model_values(device=device, region="bulk", name="z")

print "for contact drain"
for i in range(len(at_drain)):
  if at_drain[i]:
    print "%d %d %g %g %g" % (drain_ni[i], drain_ci[i], x[i], y[i], z[i])
```

chihiro0019, 2016-02-24:
Oh, I see. Thanks.

closed 2016-02-24 by chihiro0019

juanesanchez, 2016-02-24:
This is slightly more compact:
```
from ds import *
load_devices(file="mos_2d_dd.msh")

device="mymos"

contact_node_model(device=device, contact="drain", name="at_drain", equation="1")

at_drain = get_node_model_values(device=device, region="bulk", name="at_drain")
ni = get_node_model_values(device=device, region="bulk", name="node_index")
ci = get_node_model_values(device=device, region="bulk", name="coordinate_index")
x = get_node_model_values(device=device, region="bulk", name="x")
y = get_node_model_values(device=device, region="bulk", name="y")
z = get_node_model_values(device=device, region="bulk", name="z")

print "for contact drain"
for i in range(len(at_drain)):
  if at_drain[i]:
    print "%d %d %g %g %g" % (ni[i], ci[i], x[i], y[i], z[i])
```