devsim/devsim #22: Getting Error During Install on Ubuntu 14.04 LTS

sixsamuraisoldier, 2017-12-27:
Here's the error, using the scripts to install:

package_linux.sh: line 1: RC: command not found
must specify dir name

tcaduser, 2017-12-27:
thanks for finding the error.  I corrected the first line with the proper command, #!/bin/bash, in commit 867aab57c0cf.  Since it is not marked executable by default, you can either:

`
chmod +x package_linux.sh
`

and executing, or execute using:
`
bash package_linux.sh dirname
`

as shown in 

https://github.com/devsim/devsim/blob/master/scripts/build_ubuntu_14.sh

Note that you need to specify a directory name as an argument.

sixsamuraisoldier, 2017-12-27:
Thanks for the fix! 

When I try to invoke devsim by typing in 'devsim' into terminal I get "devsim: command not found"

This is after using an unofficial docker image as well. Am I doing something wrong here?

BTW, you might want to consider making an unofficial docker image.

tcaduser, 2017-12-28:
I am not aware of the unofficial docker image.  Where do you find it?

You need to either add the path to devsim to your PATH, or use the full path to the executable.

For example:

`devsim_linux/bin/devsim`

sixsamuraisoldier, 2017-12-28:
Nevermind, very dumb mistake on my end, typo in the path >.<

closed 2018-01-01 by tcaduser