devsim/devsim #64: Installation problem in Macos 10.15 version

code-whale, 2020-09-10:
Hello, I want to try to install this software in macOS 10.15 version, and I want to install it with conda, but when I type ```conda install devsim```， conda can’t find this package, can you help me solve this problem?

tcaduser, 2020-09-10:
Please download devsim from here:
https://github.com/devsim/devsim/releases/latest

and decompress the package.  There is a file in the main folder, macos.txt, which has some basic instructions.  In your conda environment, make sure you install these packages:
conda install mkl numpy

and then set the environment variable in your shell:
export PYTHONPATH=/path/to/devsim/lib

where you point the variable to the lib folder in the installation.  From the python prompt, the
import devsim

command should work.

I have not tested the system on macOS Catalina yet, so please let me know if you have any further issues.

If you have any additional usage questions, please ask them here:
https://groups.google.com/forum/#!forum/devsim

code-whale, 2020-09-11:
@tcaduser ,Thanks for your answer, I successfully ran ```import devsim``` in jupyter notebook through the above method.At present, there is no error reported when calling the module. I will ask you again if there is a problem after I study your tutorial in depth. Thank you again for your help

closed 2020-09-11 by tcaduser