devsim/devsim #66: Testing errors

thesourcerer8, 2020-10-24:
Traceback (most recent call last):
  File "info.py", line 20, in <module>
    print(devsim.get_parameter(name="info"))
UnicodeEncodeError: 'ascii' codec can't encode character '\xa9' in position 25: ordinal not in range(128)

This error comes from the value returned by get_parameter(name="info")

Traceback (most recent call last):
  File "laux1.py", line 93, in <module>
    load_devices(file="trimesh3.msh")
devsim_py3.error: Could not open file trimesh3.msh

I do not see that file there, but a trimesh2.msh

Traceback (most recent call last):
  File "mos_2d.py", line 63, in <module>
    devsim.solve(type="dc", absolute_error=1.0e30, relative_error=1e-5, maximum_iterations=30)
devsim_py3.error: Convergence failure!

Traceback (most recent call last):
  File "mos_2d_restart.py", line 16, in <module>
    devsim.load_devices(file="mos_2d_dd.msh")
devsim_py3.error: Could not open file mos_2d_dd.msh

I dont see any file like this one.

UnicodeEncodeError: 'ascii' codec can't encode character '\u03c8' in position 149: ordinal not in range(128)
The above exception was the direct cause of the following exception:

Hmm, here the stacktrace is missing.

SystemError: <built-in method flush of _io.BufferedWriter object at 0x7ff23f50da98> returned a result with an error set
The above exception was the direct cause of the following exception:
SystemError: <built-in method flush of _io.BufferedWriter object at 0x7ff23f50da98> returned a result with an error set
The above exception was the direct cause of the following exception:
SystemError: <built-in method flush of _io.BufferedWriter object at 0x7ff23f50da98> returned a result with an error set
The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "utf8_2.py", line 92, in <module>
    solve( type="dc", absolute_error= 1.0 , relative_error= 1e-10 , maximum_iterations= 30)
SystemError: <built-in function solve> returned a result with an error set

tcaduser, 2020-10-24:
To handle the © symbol in the output, it is necessary to have the locale set.  Setting:
```
export LANG=C.UTF-8
```

fixes the first issue.  I do not know the most general setting.  But this seems to be one available without having to do locale generation.  https://lintut.com/how-to-set-up-system-locale-on-ubuntu-18-04/

tcaduser, 2020-10-24:
multiprecision enabled with commit 9bdb891c4c488656174b521d9575c6b9097b3f61

tcaduser, 2020-10-24:
the recent commit also fixes utf8_2.py

tcaduser, 2020-10-24:
Some of the tests have a dependency, as can be seen in the ``CMakeLists.txt`` file in the ``testing`` directory.
```
set_tests_properties("testing/laux1" PROPERTIES DEPENDS testing/trimesh2)
```

laux1.py also needs numpy, which I have added to the build script.  so it now runs

tcaduser, 2020-10-24:
``mos_2d.py`` fails and this is probably related to using the ``SuperLU`` solver instead of ``MKL Pardiso``.  I will have to investigate this further.  ``mos_2d_restart2`` has a dependency on mos_2d.

tcaduser, 2020-10-24:
An alternative option is to do a build with ``Anaconda Python``, which is packaged with ``Intel MKL``.

thesourcerer8, 2020-10-26:
Thanks, I improved my Dockerfile accordingly.

closed 2021-06-10 by tcaduser