devsim/devsim #68: MKL can't be linked with Anaconda(Python 3.8.8)

dertyp123, 2021-06-18:
Hi@all,

i can not import DevSIM after installing "devsim_win64_v1.6.0" with pip. There is allways the following Error

`from .devsim_py3 import *
ImportError: DLL load failed while importing devsim_py3: Modul not found`

Running the "install.py " gives the following:

> Python Version 3.8.8
>
>WARNING: Intel MKL could not be dynamically loaded.
>WARNING: If you are using Anaconda/Miniconda, please issue this command:
>WARNING: conda install numpy mkl
>WARNING: mkl_rt.dll not found in CONDA_PREFIX
>
>INFO: Writing C:\Software_Script\devsim_win64_v1.6.0\lib\setup.py
>INFO:
>INFO: Please type the following command to install devsim:
>INFO: pip install -e lib
>INFO: 
>INFO: To remove the file, type:
>INFO: pip uninstall devsim

If I change Line28 in the "install.py" 
from `            'Windows' : {'mkl' : 'mkl_rt.dll'},`
to     `            'Windows' : {'mkl' : 'mkl_rt.1.dll'},`

the warning is gone.
I've figured out, the all MKL related DLL's have this *.1.dll look.

Is there a way to solve this strange linking, without copying/modifying all DLL's ?

Thx in advance for ideas and commernts
Ronny

tcaduser, 2021-06-19:
Thanks for reporting this.  It appears that Intel MKL has started versioning their libraries:
https://software.intel.com/content/www/us/en/develop/documentation/onemkl-windows-developer-guide/top/getting-started/shared-library-versioning.html

Even after fixing this script, I anticipate you will have issues because the windows dll would be hardcoded to load ``mkl_rt.dll`` instead of ``mkl_rt.1.dll``.  You can test this in python by issuing:
```
import devsim
```
at the command line.  Can you please try the following:
```
conda install mkl==2020.2 numpy
```

This will install the older version of the mkl, with ``mkl_rt.dll``.  Please let me know if this works for you.

In a future version, we will need to consider including the mkl runtime libraries as part of the distribution, or updating to the newer ``mkl_rt.1.dll`` in our next release.

If you have any general questions about using the software, please join our forum at https://forum.devsim.org

dertyp123, 2021-06-28:
Hi Tcaduser,

thx for your reply. Downgrading my Anaconda installation was not as easy as I thought.
But finally the MKL2020.2 worked. 

Thx again.

tcaduser, 2021-06-28:
You should keep a backup copy of ``mkl_rt.dll``.

While I haven't tested this, you should be able to copy the file to ``devsim_win64_v1.6.0/lib/devsim/`` so that you can use newer environments with a different mkl version.

In the next release, I'll be sure to build against ``mkl_rt.1.dll``.

closed 2021-11-25 by tcaduser