devsim/devsim #69: Error when building from source in a Ubuntu machine

AdrianTianyi, 2021-07-08:
Hi, my computer is ubuntu 18.04 and I want to build from source, though the binary version works fine. 
As shown in INSTALL, I should run  
sh scripts/build_ubuntu_18.04.sh
However, when I run this command, it tells me that external/symdiff does not appear to contain CMakeLists.txt.
If I just use cmake .,
it also fails.

So how should I build from source? Or shall I edit the .travis.yml first? If it is, how?

Thanks very much!

tcaduser, 2021-07-09:
Prerequisite packages can be installed using ``scripts/install_ubuntu_depend.sh``

The ``.travis.yml`` builds devsim on a docker centos 7 image.  However, you can use the lines in the file to build a version that should work on most modern linux distributions.  An advantage of the travis build, is that it uses the Intel MKL, as well as their linear solver, MKL Pardiso.

The ``external/symdiff`` directory is for a submodule.  To check it out:
```
git submodule init
git submodule update
```

You will also need to call this script (build_ubuntu_18.04.sh) from a working Miniconda/Anaconda environment.  Please see ``scripts/install_miniconda_linux.sh``

Once you have activated the environment, the build script should work.

tcaduser, 2021-07-09:
If you would like to start out by running the simulator, there are binary releases here:
https://github.com/devsim/devsim/releases/latest

Please note that there has been a recent change in the Intel MKL library name, so you may have to modify your Anaconda environment, as mentioned in Issue #68

AdrianTianyi, 2021-07-09:
Thanks. I got it. I run 
`git submodule init`
`git submodule update`
and get the files needed.
By the way, when calling the script in an Anaconda environment, assume I have used
`source ${HOME}/anaconda/bin/activate python3_devsim_build`
 shall I also modify build_ubuntu_18.04.sh since this script seems using python in the root directory which is not Anaconda python?

AdrianTianyi, 2021-07-09:
Let me repeat the steps to install from source.
Step 1: Install prerequisite packages listed in  scripts/install_ubuntu_depend.sh, e.g. run this script.
Step 2: Use
`git submodule init`
`git submodule update`
to get the file for symdiff.
Step 3: Use scripts/install_miniconda_linux.sh to activate an Anaconda environment.
Step 4: In this environment, run 
`sh scripts/build_ubuntu_18.04.sh`
Step 5: Run install.py to install in python.
Is that right?

AdrianTianyi, 2021-07-09:
Sorry to bother you again. When I am building files to devsim/linux_x86_65_release, I got error in building CXX object `devsim/src/MathEval/GaussFermi`
which says
`error: call to non-constexpr function ‘boost::multiprecision::number<Backend, (boost::multiprecision::expression_template_option)0> boost::multiprecision::operator*(boost::multiprecision::number<Backend, (boost::multiprecision::expression_template_option)0>&&, boost::multiprecision::number<Backend, (boost::multiprecision::expression_template_option)0>&&) [with B = boost::multiprecision::backends::float128_backend]’
     static constexpr T sqrt2_pi = boost::math::constants::root_two<T>() * boost::math::constants::one_div_root_pi<T>();
                        ^~~~~~~~
`
How could I solve this?
Thanks a lot for your kindly replys!

tcaduser, 2021-07-09:
It looks like you do not need to activate an Anaconda environment for this build.

The compiler issue is due to the Ubuntu 18 ``boost`` installation being too old to compile GaussFermi.cc, without changes.

As a workaround, please change the setting in ``linux_x86_64_release/CMakeCache.txt`` to:
```
DEVSIM_EXTENDED_PRECISION:BOOL=OFF
```

When the build is complete:
```
cd dist
bash package_linux.sh devsim_ubuntu
```

and that should create a working distribution as:
```
devsim_ubuntu.tgz
```

You should then be able to go into the ``devsim_ubuntu`` directory and type:
```
export PYTHONPATH=$PWD/lib
```

And then from a ``python3`` prompt try:
```
import devsim
```

If this works, you can try the scripts in the ``examples`` and ``testing`` directories.

The ``install.py`` script should work, if you want to use from a Anaconda python environment.

AdrianTianyi, 2021-07-10:
Thanks very much for your kindly reply. I will try this approach.

AdrianTianyi, 2021-07-12:
Thanks a lot for your patient instructions!
I have tried this approach and it works. I built it in an Anaconda environment and it seems I do not need to run install.py again.
By the way, will this building option influence the behavior of the procedure if I only care about double/single precision?

tcaduser, 2021-07-13:
I am glad you were able to get it to work.  Double precision is the default, so extended precision is not used unless enabled in the simulation script.

AdrianTianyi, 2021-07-14:
Thanks a lot for your kindly reply! I got it.

closed 2021-11-25 by tcaduser