devsim/devsim #70: interface charges

balloonk, 2021-07-15:
Is it possible to apply interface condition in the presence of interfacial charges, which according to Gauss law is
[image]
?

I read the manual about interface equation but still cannot figure it out.
It seems that interface equation supports integration of flux term but description is not very clear and I cannot find any examples for it.
I guess I have to use PotentialEdgeFlux, which is Permittivity * ElectricField, to set up the interface model?
Any help is appreciated.

fendou1997, 2021-07-15:
I have the same problem

ythran, 2021-07-15:
Hi:

Juan has created a forum for Devsim in https://forum.devsim.org/.  So we can discuss its usage in this forum.

I have a solution for the interface charge. You can create a charge node model on one side of the interface and set the concentration decreasing exponentially with depth. Then you should keep the integration of bulk charge equal to the surface charge.

Hope this will be useful.

fendou1997, 2021-07-15:
Hi:
Thanks for your reply, can you show me your code about this section?
thank you!

tcaduser, 2021-07-17:
You would need to add a node model component to the equation on one side of the interface for the potential equation.  This is similar to adding a doping to the semiconductor.  If you want to add a surface density you would need to scale it by ``SurfaceArea/NodeVolume``.  This is since the ``equation`` command will integrate the charge with respect to NodeVolume.

If you would like to discuss this further, please create a topic on the forum.

balloonk, 2021-07-18:
@tcaduser 
Thank you for the reply.
I have just implemented fixed interface charges according to your advice by creating a node solution, setting the the interface values as surface density scaled by `SurfaceArea/NodeVolume` and adding it to the RHS of Poisson equation.
Here is my implementation for users facing the same problem:
```
def set_interface_charge(device, interface, name, value):
    area_name = "SurfaceArea"
    vol_name = "NodeVolume"
    region_list = devsim.get_region_list(device=device, interface=interface)
    for region in region_list:
        values = list(devsim.get_node_model_values(device=device, region=region, name=name))
        edge_list = devsim.get_element_node_list(device=device, region=region, interface=interface)
        node_list = numpy.unique(numpy.array(edge_list, int).flatten())
        surf_area = devsim.get_node_model_values(device=device, region=region, name=area_name)
        node_vol = devsim.get_node_model_values(device=device, region=region, name=vol_name)
        for node in node_list:
            values[node] = value * surf_area[node] / node_vol[node]
        devsim.set_node_values(device=device, region=region, name=name, values=values)
```
```
devsim.node_solution(device="device", region=region1, name="InterfaceCharge")
devsim.node_solution(device="device", region=region2, name="InterfaceCharge")
set_interface_charge("device", interface_name, "InterfaceCharge", 1e10)
RHS = "-ElectronCharge*kahan4(Holes, -Electrons, NetDoping, InterfaceCharge)"
```

fendou1997, 2021-07-20:
Can the interface charge density be set time-varying? For example, the charge density is about the electric fields at the interface.

tcaduser, 2021-07-22:
You would need a model for interface charge density with respect to other solution variables.  You can also add a time dependent trap continuity equation.  A dependence on electric field could potentially be done with an edge or element edge model, as described here:
https://doi.org/10.1109/TED.2021.3094776

Do you have a specific model in mind?

fendou1997, 2021-07-22:
I have a special interface model whose equation is different from the bulk equation. In my application, what's more, the equation at the two side of interface is different.  In my mind, the quantity of interface conditions for a two order partial differential equation should 2 conditions. In your software, I only find a condition setup, I guess the another condition is integrated by the bulk equation. So my mind is right? Can you have the solution to solve my demand? Thank you!

tcaduser, 2021-11-25:
I am sorry if I did not follow up on this issue with you before.  I hope you found a solution.

closed 2021-11-25 by tcaduser