devsim/devsim #71: the questions about the solver transient

fendou1997, 2021-07-15:
I want to calculate the transient distribution of MOS structure with the gate voltage is applied the square wave? can you show me an example to achieve it? The corresponding example in this project is no about it. I want to known how to set the gate voltage? Thank you!

tcaduser, 2021-07-17:
As you have observed, there has not been a lot of development with the transient solver, so please consider it to be experimental.

Since the ``solve`` command does only one time step at a time, you would need to control the time stepping yourself.  The gate bias would need to be set appropriate value at each time step.  For a square wave, there would be a break point for each corner of the wave form.

Please create a topic at:
https://forum.devsim.org
if you wish to discuss this further.

tcaduser, 2021-11-25:
The upcoming release 1.6.1 will have some updates to concerning the transient solver.  Please track the forum for updates.

closed 2021-11-25 by tcaduser