devsim/devsim #72: The generalization of the example diode in the package

AdrianTianyi, 2021-07-20:
Hi, I am now using DEVSIM and have read the documents and examples therein. It looks pretty well. However, I have the following problems in the example diode:
1. Do we have a way to obtain a transient solution of diode in a certain time? If we cannot, does the final solution (stored in data variable) in the example guaranteed to be the steady state dc solution?
2. If I want to solve another similar problem with different geometry shape in 2D, how should I modify the script? From my understanding, all I need is to modify the codes for creating mesh, setting parameters and setting net doping. Then I will obtain the steady state dc solution using the script for diode example. 
Thanks a lot!

tcaduser, 2021-07-22:
As mentioned in Issue #71, the solver does a single time step per transient solve.  It would be necessary to have a loop in your Python script which changes the bias with respect to time and then track the response.

```
examples/diode/diode_2d.py
examples/diode/gmsh_diode2d.py
```

use the same physics, starting with a different structure.  The first uses the internal mesher.  The second example uses a mesh generated from ``Gmsh`` and loaded in the simulator.

If possible, please consider using the forum to continue this discussion:
https://forum.devsim.org

AdrianTianyi, 2021-07-22:
Thanks a lot for your reply. In fact I am now testing a modified version of diode_2d.py, but wondering what's the result I obtain, i.e., transient solution or steady solution. (Note that tdelta is set to 0 by default.)
  I will put this question on https://forum.devsim.org.

closed 2021-11-25 by tcaduser