devsim/devsim #77: [JOSS Review] Instructions for using MacOS binaries need update

dalonsoa, 2021-11-24:
**Part of the review for publication in [JOSS](https://github.com/openjournals/joss-reviews/issues/3898)** 

When following the instructions to setup `devsim` binaries on MacOS, I get two issues that should be described in the instructions:

When doing:

```python
import devsim
```

I get a message saying that the `.so` file is not authorised. I need to authorize the use of libraries from unknown developers (at least in MacOS Monterrey 12.01). This is similar to the issue - and solution - described [here](https://github.com/SAP/node-rfc/issues/140#issuecomment-675260978). 

After doing that, then it complains because MKL is not found. 

```output
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/Users/dalonsoa/Documents/REVIEWS/devsim/macos/lib/devsim/__init__.py", line 8, in <module>
    from .devsim_py3 import *
ImportError: dlopen(/Users/dalonsoa/Documents/REVIEWS/devsim/macos/lib/devsim/devsim_py3.so, 0x0002): Library not loaded: @rpath/libmkl_rt.dylib
  Referenced from: /Users/dalonsoa/Documents/REVIEWS/devsim/macos/lib/devsim/devsim_py3.so
  Reason: tried: '/usr/lib/libmkl_rt.dylib' (no such file)
```

I believe [instructions for MacOS](https://github.com/devsim/devsim/blob/c6ee02f1ab553f65754752c370ae3dc1c1548224/macos.txt) are slightly misleading. Having MKL **is required**, not optional as it is sugetsed in the documentation, which states that "one of these packages" in reference to Python 3 and Intel MKL. Both are needed. 

Once MKL is installed, all works as expected. 

Also, it will be nice if the different binary installation files were in Markdown or other format that renders nicely.

tcaduser, 2021-11-25:
Hello @dalonsoa,
Thank you for your comments.  I have updated the instructions for installing the binaries in INSTALL.md for commit 52e575d606e.  This now includes the requirement for Intel MKL.

My mac laptop is based on Mojave (Version 10.14.6).  I am not familiar with this restriction for modules loaded via Anaconda.  I have seen this issue for modules loaded via the system Python 2.7 interpreter in Mojave, since it is an executable attempting to load a shared library in a non-system directory.

Is your Anaconda installation located in your home directory, or in a system directory?  At the moment, I do not have access to Monterrey 12.01 system.

dalonsoa, 2021-11-25:
Yes, `conda` is in my home director (in `~/opt/miniconda3/bin/conda`, in particular).

Who knows about the restriction! Apple adds more and more restrictions with each release of MacOS. I am not surprised at all that it works in Mojave but not in Monterry. I would mention in the installation instructions that this issue *might* hapen, as well as a possible solution. It could be in a Troubleshoot section or something like that.

tcaduser, 2021-11-26:
I will add a note to the INSTALL.md

tcaduser, 2021-11-26:
please see commit f1518377a

closed 2022-01-11 by tcaduser