devsim/devsim #78: [JOSS Review] Contributing Guidelines are outdated

dalonsoa, 2021-11-24:
Part of the review for publication in [JOSS](https://github.com/openjournals/joss-reviews/issues/3898)

The [contributing guidelines](https://github.com/devsim/devsim/blob/c6ee02f1ab553f65754752c370ae3dc1c1548224/CONTRIBUTING.md) need to be updated - and possibly expanded with more informatin. They point to a [mailing list](https://groups.google.com/g/devsim?pli=1), but that one was closed in April and now a [Forum](https://forum.devsim.org/) must be used. 

They are also quite short and limitted, and not asnwering explicitly how to:
1. Contribute to the software
2. Report issues or problems with the software
3. Seek support

All they say is "go to the forum". 

I would say that using a forum is not too standard for open source projects - using the issue tracker is more normal - but if it works withinn that communnity, I guess it is fine.

tcaduser, 2021-11-25:
Hi @dalonsoa,

Thank you for your comments.  With commit 52e575d606e, I have updated CONTRIBUTING.md to point to the new forum.  My understanding of a contributing file, is that it is for new development.  I was not aware that it is instructions concerning support.  Should I be moving the Support section of README.md into CONTRIBUTING.md?  Your help is greatly appreciated.

dalonsoa, 2021-11-25:
You are absolutely right: The contributing file is about how to contirbute to the software. For support, asking to open an issue - or go to the forum in your case - is enough. 

Nevertheless, I do believe the CONTRIBUTING file needs improvement to help possible contirbutors learn what tools should be used, how the code should be structured, etc. For example, what are the expectations about testing new code? Or what should a new feature be placed? They cal always ask in the forum, but it helps if that information is offered upfront. You can have a look at the [CONTRIBUTING file for the PyBaMM project](https://github.com/pybamm-team/PyBaMM/blob/develop/CONTRIBUTING.md) (I feel that one is excessive, to be honest, but it is a good example).

Now, I see you are, pretty much, the sole contributor to the project. If you want to keep it that way for waheter reason, that is absolutely fine, but then you must make that clear: that you do not accept external contributors. There is noting wrong about that - `sqlite` for example follows that model; check [section about "Open-source, not Open-contribution" in their copyrigth page](https://sqlite.org/copyright.html). But it has to be clear upfront.

tcaduser, 2021-11-26:
@dalonsoa 

I welcome contributions, but since there have been few, the process has not been formalized.

Most of the contributions to the software has been through bug reporting.  Some users of the software have shared their physics models or simulation structures in Python as independent projects, or documented them as journal papers.  I refer to them on the project webpage as I am made aware of them.
https://devsim.org/introduction.html#using-the-simulator

As far as changes to the core application, I have had 2 people contribute build scripts for their platform and 1 person provide a patch for a newer version of SuperLU.  One person provide a python prototype for the "Gauss Fermi" integral, which I translated to C++.

closed 2022-01-11 by tcaduser