devsim/devsim #88: Deleting meshes and devices?

mdecea, 2022-09-12:
I have a function that repetitively creates a device (let's say a 1D diode) with varying parameters. Since this is a function, it tries to generate meshes ("dio") and devices ("MyDevice") with the same name every time I call the function. 

Nevertheless, this seems to not work and I get the following error:  `devsim_py3.error: A mesh already exists with name "dio"`.
I cannot reuse the already existing mesh because some of the dimensions changed. Is there a way to avoid this behavior? Maybe a way to delete the previous mesh (and device) so that this is not an issue?

Thanks!

tcaduser, 2022-09-12:
Hi @mdecea,

In principal, this could be fairly straightforward, as the data structures were designed to be flexible.

Would you be willing to provide a failing test case, so I can understand your use case?

As a temporary workaround, perhaps you can execute a new python process for each new device?  Alternatively, you can have a whole bunch of devices being solved simultaneously by instantiating devices with different names.

I'm thinking I could get this out as part of a new release within the next few weeks.

mdecea, 2022-09-13:
Hi @tcaduser,

Thanks for your fast reply.

Here is a failing test case:

import devsim
from devsim.python_packages import model_create, simple_physics

```
um = 1e-6

class Diode_1D():

    """1D pn junction model. 

    Parameters:
        p_width: width of the p region
        p_conc: lacceptor concentration (/cm3).
        n_conc: ldonor concentration (/cm3).
        atol: tolerance for iterative solver.
        rtol: tolerance for iterative solver.
        max_iter: maximum number of iterations of iterative solver.
    """

    def __init__(self, p_width=0.5e-5, p_conc=1e18, n_conc=1e18, atol=1e6, rtol = 1e-12, max_iter=60):
        self.p_width = p_width
        self.p_conc = p_conc
        self.n_conc = n_conc
        self.atol = atol
        self.rtol = rtol
        self.max_iter = max_iter
        self.total_width = 1e-5
    
    def create_1d_mesh(self, device):

        # ----- Mesh creation ----
        devsim.create_1d_mesh(mesh="dio")

        devsim.add_1d_mesh_line(mesh="dio", pos=0, ps=1e-7, tag="top")
        devsim.add_1d_mesh_line(mesh="dio", pos=self.p_width, ps=1e-9, tag="mid")
        devsim.add_1d_mesh_line(mesh="dio", pos=self.total_width, ps=1e-7, tag="bot")

        # ---- Contacts ----
        devsim.add_1d_contact  (mesh="dio", name="top", tag="top", material="metal")
        devsim.add_1d_contact  (mesh="dio", name="bot", tag="bot", material="metal")

        # ---- Regions ----
        devsim.add_1d_region   (mesh="dio", material="Si", region="full", tag1="top", tag2="bot")

        devsim.finalize_mesh(mesh="dio")
        devsim.create_device(mesh="dio", device=device)

    def set_parameters(self, device) -> None:
        """Set parameters for 300 K."""
        simple_physics.SetSiliconParameters(device, "full", 300)

    def set_net_doping(self, device) -> None:
        """NetDoping."""

        # P doping 
        model_create.CreateNodeModel(
            device, "full", "Acceptors", 
            f"{self.p_conc:1.3e}*step({self.p_width:1.3e}-x)",
        )

        # N doping
        model_create.CreateNodeModel(
            device, "full", "Donors", 
            f"{self.n_conc:1.3e}*step(x-{self.p_width:1.3e})",
        )

        model_create.CreateNodeModel(device, "full", "NetDoping", "Donors-Acceptors")

    def initial_solution(self, device, circuit_contacts=None):

        # Create Potential, Potential@n0, Potential@n1
        model_create.CreateSolution(device, "full", "Potential")

        # Create potential only physical models
        simple_physics.CreateSiliconPotentialOnly(device, "full")

        # Set up the contacts applying a bias
        for i in devsim.get_contact_list(device=device):
            if circuit_contacts and i in circuit_contacts:
                simple_physics.CreateSiliconPotentialOnlyContact(device, "full", i, True)
            else:
                ### it is more correct for the bias to be 0, and it looks like there is side effects
                devsim.set_parameter(device=device, name=simple_physics.GetContactBiasName(i), value=0.0)
                simple_physics.CreateSiliconPotentialOnlyContact(device, "full", i)

    def drift_difussion_initial_solution(self, device, circuit_contacts=None):
        """ drift diffusion solution variables
        """
        
        model_create.CreateSolution(device, "full", "Electrons")
        model_create.CreateSolution(device, "full", "Holes")

        #create initial guess from dc only solution
        devsim.set_node_values(device=device, region="full", name="Electrons", init_from="IntrinsicElectrons")
        devsim.set_node_values(device=device, region="full", name="Holes",     init_from="IntrinsicHoles")

        # Set up equations
        simple_physics.CreateSiliconDriftDiffusion(device, "full")
        for i in devsim.get_contact_list(device=device):
            if circuit_contacts and i in circuit_contacts:
                simple_physics.CreateSiliconDriftDiffusionAtContact(device, "full", i, True)
            else:
                simple_physics.CreateSiliconDriftDiffusionAtContact(device, "full", i)

    def ddsolver(self) -> None:
        """Initialize mesh and solver."""
        device = "MyDevice"
        self.create_1d_mesh(device)
        self.set_parameters(device=device)
        self.set_net_doping(device=device)

        self.initial_solution(device=device)

        # Initial DC solution
        self.drift_difussion_initial_solution(device=device)

        for interface in devsim.get_interface_list(device=device):
            simple_physics.CreateSiliconSiliconInterface(
                device=device, interface=interface
            )

        devsim.solve(
            type="dc",
            absolute_error=self.atol,
            relative_error=self.rtol,
            maximum_iterations=self.max_iter,
        )

if __name__ == "__main__":

    c = Diode_1D(p_width=0.5e-6)
    c.ddsolver()

    c = Diode_1D(p_width=0.6e-6)
    c.ddsolver()
```

Executing this code will throw a `devsim_py3.error: A mesh already exists with name "dio"`. 

This is a simple example, but the use case I'm thinking of is when trying to optimize a structure. I am trying to use some optimizer (for example from `scipy`). While I can of course write the function to optimize so that it generates  a different mesh name and device name, I think ideally there should be a way to "erase" the device and reuse the names.

mdecea, 2022-09-13:
An addition: I have implemented the temporary workaround where the different device variants are generated with a different mesh and device name. 

When I execute the code I see something interesting: when trying to solve the drift-diffusion equations on a given device, it looks like the solver is actually solving all the existing devices and not only the one I'm interested in. 

I'm saying this because the info printed by the solver includes all the devices I have generated. If that's the case I think this should be fixed.

Here is an example output:

...
```
 (region: full)
 (contact: p_contact)
 (contact: n_contact)
Warning: Replacing equation with equation of the same name.
Region: full, Equation: PotentialEquation, Variable: Potential
number of equations 2298
Iteration: 0
  Device: "dev1"    RelError: 2.53807e-14   AbsError: 3.76940e+04
    Region: "full"      RelError: 2.53807e-14   AbsError: 3.76940e+04
      Equation: "ElectronContinuityEquation"    RelError: 1.05622e-15   AbsError: 8.03528e+03
      Equation: "HoleContinuityEquation"        RelError: 1.53650e-15   AbsError: 2.96587e+04
      Equation: "PotentialEquation"     RelError: 2.27880e-14   AbsError: 5.35227e-17
  Device: "dev2"    RelError: 2.02000e+10   AbsError: 2.02000e+20
    Region: "full"      RelError: 2.02000e+10   AbsError: 2.02000e+20
      Equation: "ElectronContinuityEquation"    RelError: 1.01000e+10   AbsError: 1.01000e+20
      Equation: "HoleContinuityEquation"        RelError: 1.01000e+10   AbsError: 1.01000e+20
      Equation: "PotentialEquation"     RelError: 1.00000e+00   AbsError: 1.09336e-01
Iteration: 1
  Device: "dev1"    RelError: 4.17634e-14   AbsError: 3.67919e+04
    Region: "full"      RelError: 4.17634e-14   AbsError: 3.67919e+04
      Equation: "ElectronContinuityEquation"    RelError: 1.62416e-15   AbsError: 8.03575e+03
      Equation: "HoleContinuityEquation"        RelError: 8.07786e-16   AbsError: 2.87562e+04
      Equation: "PotentialEquation"     RelError: 3.93314e-14   AbsError: 5.35251e-17
  Device: "dev2"    RelError: 2.27503e+03   AbsError: 1.56074e+20
    Region: "full"      RelError: 2.27503e+03   AbsError: 1.56074e+20
      Equation: "ElectronContinuityEquation"    RelError: 1.26588e+03   AbsError: 7.73087e+19
      Equation: "HoleContinuityEquation"        RelError: 9.99000e+02   AbsError: 7.87653e+19
      Equation: "PotentialEquation"     RelError: 1.01517e+01   AbsError: 7.86617e-02
```
...

I'm trying to solve only dev2, but it seems like both dev1 and dev2 are being solved.

tcaduser, 2022-09-14:
Hi @mdecea 

The software solves all of the devices simultaneously, including any boundary conditions that may be connecting them.  I think the appropriate fix is to implement the mesh and device deletion commands you requested.

Thank you for the test case.  I will try to get this feature implemented soon.

tcaduser, 2022-09-17:
Please try out the new 2.2.0 release and see if the `delete_device` and the `delete_mesh` commands meet your requirements.

tcaduser, 2022-09-17:
```
    c = Diode_1D(p_width=0.5e-6)
    c.ddsolver()

    devsim.delete_mesh(mesh="dio")
    devsim.delete_device(device="MyDevice")
    c = Diode_1D(p_width=0.6e-6)
    c.ddsolver()

    devsim.delete_mesh(mesh="dio")
    devsim.delete_device(device="MyDevice")
    c = Diode_1D(p_width=0.5e-6)
    c.ddsolver()
```

mdecea, 2022-09-20:
Works great! Thanks.

Closing the issue.

closed 2022-09-20 by mdecea