devsim/devsim #89: pip install devsim

joamatab, 2022-09-15:
How can we make devsim available in PyPi?

it would make the installation much easier in the instructions for the [gdsfactory plugin](https://gdsfactory.github.io/gdsfactory/notebooks/plugins/devsim/01_pin_waveguide.html)

@proppy
@mithro

simbilod, 2022-09-16:
also see https://github.com/hdl/conda-eda/issues/240

proppy, 2022-10-11:
I'd encourage you to take a look at https://github.com/pypa/cibuildwheel which could help to put together the automation to build multiple platform wheels.

tcaduser, 2022-10-12:
this has truly been a dreadful exercise.  exploring using the wheel package.
```
# setup.cfg
[bdist_wheel]
py-limited-api = cp36
```

```
#setup.py
from setuptools import setup, Extension

module1 = Extension('devsim.foo',
                    py_limited_api = True,
                    sources = ['foo.c'])
setup(name='devsim',
      version = '1.0.0',
      packages=['devsim', 'devsim.python_packages'],
      package_dir = { '' : 'lib'},
      package_data={'devsim' : ['./*.so',]},
      ext_modules = [module1],
      )
```

and using the command line:
```
python setup.py bdist_wheel
```

I am able to create a wheel package from an existing `devsim` distribution.  I still need to find the example files.  The dummy `foo` module is unfortunate, and I may have to use trickery to remove it from the final distribution.

To summarize, python packaging is terrible.

tcaduser, 2022-10-14:
I have create an automated build with binary wheels for the following intel 64 bit platforms:
macOS, Linux, Windows.

They should be able to be installed using this command:
```
pip install -i https://test.pypi.org/simple/ devsim
```
On a conda system, the example files can be tested here:
```
cd ${CONDA_PREFIX}/devsim_data/testing
python cap2.py
```

Please note that an MKL installation is required.
```
conda install mkl
```

If someone could try this, it would be greatly appreciated.

If the pypi test server has been purged, the wheel files are available here:
https://github.com/devsim/devsim/releases/tag/vbdist6

simbilod, 2022-10-14:
The examples ran for me on both Ubuntu and Windows after creating a new conda environment. 

I also tried with `pip install mkl` which also worked, so hopefully this also works in regular non-conda venvs.

Is the https://test.pypi.org/simple/ URL just a placeholder?

tcaduser, 2022-10-14:
@simbilod 
Thank you for trying the examples out.  The test.pypi.org server is a meant for testing deployments.  The packages can be purged at any time.

I intentionally left out any dependencies since MKL is huge, and zlib and sqlite are part of the Python standard library.

Are there any major platforms missing, or is intel 64 bit on Linux, macOS, Windows sufficient?

joamatab, 2022-10-20:
Hi Juan,

let us know when you push to pypi so we can simplify the devsim installation for gdsfactory

tcaduser, 2022-10-20:
Hi @joamatab 

It should be done by this weekend.  I will  update this ticket accordingly.

tcaduser, 2022-10-21:
Hi @joamatab,

I have created the 2.3.0-rc2 release, and it can be installed using:
```
pip install --pre
```

One remaining issue is that the MKL is not found in the Ubuntu 22 python 3 environment, unless I set the LD_LIBRARY_PATH to ``$VIRTUAL_ENV/lib``.  Do you know if there is a way to set the environment in this situation?

I am still on track for a release this weekend, after updating the documentation.

tcaduser, 2022-10-22:
Hi @joamatab , I have made a release.  Please let me know if you encounter any issues.

tcaduser, 2022-10-22:
while this release is a good start, I still need to work at getting the directory structure to be properly respected  in `devsim_data`.  globbing files results in files all in the same directory

tcaduser, 2022-10-22:
This appears to be fixed in 2.3.2, aa7e166e5c

joamatab, 2022-10-22:
Hi Juan,

`pip install devsim` works great on linux

tcaduser, 2022-10-22:
Thanks @joamatab for testing it out.  The latest patch releases are just to fix the metadata and the relative links on the pipy project page.

closed 2022-10-28 by joamatab