devsim/devsim #92: diode1d and diode2d results not consistent

mdecea, 2022-09-20:
I'm trying to get a diode simulation working based on the example given in `devsim/examples/diode`, but I'm getting results that are a bit confusing:

- I get a different current for the p contact and for the p contact (when of course these should be the same).
- In reverse bias, the results given by the 2D simulation and the 1D simulation are significantly different:

[image]

As far as I can tell both structures are analogous so I would expect the IV curve to look the same. In the plot above, I have scaled the 1D results by the height of the 2D diode (10 um). 

Any clue what might be happening?

Thanks!

tcaduser, 2022-09-21:
Ideally, I think it is best to explore these types of questions at https://forum.devsim.org, since it promotes discussion of simulation issues other people may be having.

The noisy reverse bias may possibly be alleviated by using extended precision.  This may be enabled using:
```
devsim.set_parameter(name = "extended_solver", value=True)
devsim.set_parameter(name = "extended_model", value=True)
devsim.set_parameter(name = "extended_equation", value=True)
```
This feature is available on all platforms.  It is new on windows starting with release 2.2.0.

Hopefully this results in better current continuity between the n and p contacts.  If it does not, then this may be a bug.  Please ensure your simulations are running to a relative error on the order of 1e-12 or smaller.  For extended precision, this should be attainable.

Since the place where recombination is occurring may change with reverse bias, it may be necessary to refine the mesh over a wider area of the device.  I do have some adaptive meshing scripts for a gmsh created mesh here:
https://github.com/devsim/devsim_misc/tree/main/refinement

An easier approach is to make the mesh finer as part of your initial device creation.

tcaduser, 2023-01-25:
Closing based on how long this has been open.  Please reopen if this is still an issue for you.

closed 2023-01-25 by tcaduser