devsim/devsim #93: The calculation of heat generation

ZhenglaiT, 2022-10-26:
Hello everyone!
I am trying to calculate the heat generation and temperature distribution. Furthermore, I want to consider the  simulation calculation of thermoelectric coupling. 
Currently, I am calculating the temperature field considering only Joule heating based on the example of a 2D MOSFET. The control equation are below:
[image]
First, to make sure that my code assembles the equations correctly, I first set up a uniform heat source for the calculations. Attached is my modified code.But I meet some problems. 

First, considering the suggestion of Juan , I set the parameter like mu_n, mu_p and V_t to be temperature dependence. Originally, I use the node model to describe them. But to calculate the current, it needs the edge value, so I use the function of "EnsureEdgeFromNodeModelExists". The error appears:
[image]
So I built an edge model of these parameters using the average temperature of the nodes, like:
`def CreateSiliconDriftDiffusion(device, region):
   
    # mu_n_T = "mu_n*pow(Temperature/T,-2.2)"
    # mu_p_T = "mu_p*pow(Temperature/T,-2.2)"
    # V_t_T = "boltzmannConstant*Temperature/ElectronCharge"
    mu_n_T = "mu_n*pow((Temperature@n0+Temperature@n1)/(2*T/100),-2.2)"
    mu_p_T = "mu_p*pow((Temperature@n0+Temperature@n1)/(2*T/100),-2.2)"
    V_t_T = "boltzmannConstant*(Temperature@n0+Temperature@n1)/(2*ElectronCharge)"
    for i in (
        ("Nmobility", mu_n_T),
         ("Pmobility", mu_p_T),
         ("VtT", V_t_T),
    ):
        n = i[0]
        e = i[1]
        # CreateNodeModel(device, region, n, e)
        # CreateNodeModelDerivative(device, region, n, e, "Temperature")
        CreateEdgeModel(device, region, n, e)
        CreateEdgeModelDerivatives(device, region, n, e, "Temperature")
    CreatePE(device, region)
    CreateBernoulli(device, region)
    CreateSRH(device, region)
    CreateECE(device, region)
    CreateHCE(device, region)
`
I don't know if it is right, but it works for now.

The most confused questions it that the absolute value of the temperature setting affects the convergence of the results. If I set the boundary or initial temperature as a few hundred kelvins, the error of "Convergence failed“ will appear. However,  If I set the boundary or initial temperature as a few kelvins. Like :
`    set_parameter(device=device, name="drain_biasT", value=3)
    set_parameter(device=device, name="source_biasT", value=3)
    set_parameter(device=device, name="gate_biasT", value=3)
    set_parameter(device=device, name="body_biasT", value=2)`
I can get the results of temperature:
[image]
From the results of temperature, the constructed heat conduction equation works. But it also brings a new problem, that is, the uniform heat source I set doesn't seem to play a role in the calculation.

In short, my questions is:
1. Why do edge models built from node models of V_t parameters fail ?
2. Why does the absolute value of temperature affect the convergence of the equation？
3. Why the heat source settings are not working？
I'm sorry I have so many questions, I‘d appreciate that if someone could give me some advice. The code I have modified is in the attachment.
[Heat_generation calculation.zip](https://github.com/devsim/devsim/files/9865572/Heat_generation.calculation.zip)

ZhenglaiT, 2022-10-26:
I'm sorry that my heat source setting has an effect on the temperature distribution in fact, but the value should not be too large to avoid data overflow

ythran, 2022-10-26:
I think Github is not the best place to discuss how to use Devsim. You can migrate this question to the forum of Devsim
https://forum.devsim.org/

ZhenglaiT, 2022-10-27:
Hi ythran 
I am sorry for this. I create this problem in the forum originally. But Juan told me to create an issue in the github. I will be careful in the next time. Thank you for your suggestion.

 

 https://forum.devsim.org/

tcaduser, 2022-10-28:
Hi @ZhenglaiT 

Please try out this package of files.  If you want to understand my changes, please discuss them on the forum.
[Heat_generation-first.zip](https://github.com/devsim/devsim/files/9888884/Heat_generation-first.zip)

The modified module files are now located in the `tphysics` subdirectory.

tcaduser, 2022-11-16:
Hi @ZhenglaiT 

Thanks for sharing your work:
[https://github.com/ZhenglaiT/MOSFET-simulation](https://github.com/ZhenglaiT/MOSFET-simulation)

closed 2022-11-16 by tcaduser