devsim/devsim #94: Reduce data in VTK or TecplotFormat

tcaduser, 2022-10-30:
**Before opening an issue, please create a discussion concerning your request at** https://forum.devsim.org

**Is your feature request related to a problem? Please describe.**
The data files are huge

**Describe the solution you'd like**
Way to specify fields to be written to the output file.  Probably include a regex for models to be ignored.

**Describe alternatives you've considered**
N/A

**Additional context**
There are may fields which do not even make sense to be in a data file.  The following data is probably not useful.
* Edge flux data
* Derivative data

simbilod, 2023-01-16:
Would filtering the NodeModel and EdgeModel lists [around here](https://github.com/devsim/devsim/blob/226631ce050764074dbbb8e92c7766d116681ab8/src/meshing/TecplotWriter.cc#L444) according to a provided array of strings (corresponding to the models that are to be saved) do the trick?

On a semi-related note, is there a reason beyond history that high-level operations like file IO are implemented in C++? Some other packages like [MEEP ](https://github.com/NanoComp/meep) have made moves recently to keep only the core loops low-level, and move everything else to higher level (Python), which greatly lowers the barrier for user contributions. But I understand this could be a large refactoring.

tcaduser, 2024-06-05:
This will be addressed in the 3.8.0 release.

closed 2024-06-05 by tcaduser