devsim/devsim #97: Contact definition error message always refers to contacts needing to be 2D

simbilod, 2022-11-14:
**This template is for bug reporting. For general questions about the simulator, please create a new discussion at** https://forum.devsim.org

**Describe the bug**
In a 2D simulation using a gmsh 2D mesh, defining a 2D region for contact produces an error with message `Contact <contact name> region name <region name> must be 2 dimensional to be a contact.`

**To Reproduce**
See above

**Expected behavior**
From the example at `https://github.com/devsim/devsim/blob/d970ec9c55de6739a28a040183a5e5b152f8c068/examples/mobility/` where the drain and source contacts are 1-dimensional objects according to the msh file, it seems that the error message should promp the user to create a 1D contact if the simulation is in 2D. 

A generic error message mentionning contacts needing to be "N-1"-dimension could do.

**Screenshots**
N/A

**Desktop (please complete the following information):**
 - Debian

**Additional context**
Add any other context about the problem here.

tcaduser, 2022-11-27:
Hi @simbilod 

Thanks for reporting this.  I think this may be straightforward and will be fixed in a future release.

tcaduser, 2023-01-16:
This should be fixed with release 2.3.7

closed 2023-01-16 by tcaduser