devsim/devsim #100: Custom registered functions don't work with extended precision

Ryan3141, 2023-01-25:
**Describe the bug**
devsim is unable to find a custom function registered in python.

**To Reproduce**
Solve a system with devsim with extended precision and a custom function registered with `devsim.register_function`.

**Expected behavior**
Registered functions to behave identically under any numerical precision.

**Desktop (please complete the following information):**
Operating System: Manjaro Linux
KDE Plasma Version: 5.26.4
KDE Frameworks Version: 5.101.0
Qt Version: 5.15.7
Kernel Version: 6.1.1-1-MANJARO (64-bit)
Graphics Platform: X11

**Additional context**
In ModelCommands.cc there is a line:
`MathEval<double>::GetInstance().AddTclMath(name, procedure, static_cast<size_t>(nargs), errorString);`

This seems to not be used when extended precision is chosen in the python code, i.e:
```
devsim.set_parameter(name="extended_solver", value=True)
devsim.set_parameter(name="extended_model", value=True)
devsim.set_parameter(name="extended_equation", value=True)
```

I have had success by simply adding an additional line right next to it though I'm not sure if I should optionally choose only one:
`MathEval<float128>::GetInstance().AddTclMath(name, procedure, static_cast<size_t>(nargs), errorString);`

tcaduser, 2023-01-25:
Hi @Ryan3141,

I was able to replicate your issue by placing:
```
set_parameter(name="extended_model", value=True)
```

at the top of ``testing/testfunc.py``

```
while evaluating node model gee on Device: MyDevice on Region: MyRegion
could not find function "gee"
while evaluating model gee: expression gee(x, 1) evaluates to invalid
Traceback (most recent call last):
  File "/home/juan/git/devsim/testing/testfunc.py", line 59, in <module>
    print_node_values(device=device, region=region, name="gee")
devsim_py3.error: DEVSIM FATAL
```

Temporary workaround, disable extended model precision when defining a model using a registered function.
```
set_parameter(name="extended_model", value=False)
node_model(device=device, region=region, name="gee", equation="gee(x, 1)")
set_parameter(name="extended_model", value=True)
```

This is a bug that can be fixed in the next release.

## proposed fix
As with your solution, the evaluation of the Python function will be called using double precision, even from extended precision mode.  The double precision result will then be used in the extended precision model.  If you want to be able to evaluate python functions in extended precision, please file an enhancement request.  (_At the moment, I think the portable way would be for the python function to accept a string representation of a number_).

tcaduser, 2023-01-25:
Hi @Ryan3141 

Your solution makes sense.  Please file a pull request against branch  r2.4.0, and I'll accept it.

Thanks,

Juan

Ryan3141, 2023-01-25:
Thanks for the quick response!  I really appreciate the help with such a useful program.  This is actually my first real pull request, so let me know if I screwed anything up.  #102

closed 2023-01-25 by Ryan3141

tcaduser, 2023-01-25:
Thank you for finding the issue and providing a fix.  I put the pull request on the r2.4.0 branch, and I hope to get a new release out soon.