devsim/devsim #144: DevSim does not converge when trying 1 Ampere test for a simplest device structure

Alex236842, 2024-09-06:
**Background**
There is a one-ampere test formulated in the forum:
https://forum.devsim.org/t/drift-diffusion-simulation-for-3d-produces-higher-currents-than-expected/244/1
The device for the 1 Ampere test in 3D version has a shape of a silicon cube with two contacts on the opposite cube sides.
The silicon sample is uniformly n-doped so that the sample has a resistivity of 1 Ohm.
When a 1 Volt bias is applied to this device, the resulting current is expected to be 1 A.
In 3D device simulations presented in the forum, the resulting current was found to be higher than 1 A by about 26%.
The bottom line of the forum discussion was: this is an effect of non-delaunay mesh elements.

**Description of the bug**
In this example, I present a primitive device with a 100% Delaunay mesh for which DevSim cannot achieve convergence.
Commercial device simulators solve this test easily and with a high accuracy.

**To Reproduce**
-- copy all the files attached to a working directory from which devsim is achievable
-- Start test:
python test1a.py 
I used python 3.7, for example using "conda activate env37"
and DevSim version 2.7.0

**Expected behavior**
At the end of test output should be something like this: 
Solution:
Contact   Voltage        Electron    Hole        Total     
                         Current     Current     Current
bot       +0.000e+00	-1.001e+00	-2.878e-23	-1.001e+00
top       +1.000e+00	+1.001e+00	+2.878e-23	+1.001e+00

**Screenshots**
Here is a figure showing the device mesh used in this test
[image]

**Desktop (please complete the following information):**
 - OS: Linux 
 - Version 5.4.0-193-generic #213-Ubuntu 

**Additional context**
Files:  readme.txt  
          test1a.py                     python script to reproduce the bug
          diode_common.py      additional procedures for the test 
          gmsh_diode3d.msh    input gmesh file presenting device structure
          test1a.out                   DevSim text output demonstrating convergence failure
          gmsh_diode3d_out.msh      DevSim structure output
          gmsh_nVolume_efield.dat    DevSim structure output 
[diode_common.py.gz](https://github.com/user-attachments/files/16907066/diode_common.py.gz)
[gmsh_diode3d.msh.gz](https://github.com/user-attachments/files/16907067/gmsh_diode3d.msh.gz)
[gmsh_diode3d_out.msh.gz](https://github.com/user-attachments/files/16907068/gmsh_diode3d_out.msh.gz)
[gmsh_nVolume_efield.dat.gz](https://github.com/user-attachments/files/16907069/gmsh_nVolume_efield.dat.gz)
[readme.txt.gz](https://github.com/user-attachments/files/16907071/readme.txt.gz)
[test1a.out.gz](https://github.com/user-attachments/files/16907072/test1a.out.gz)
[test1a.py.gz](https://github.com/user-attachments/files/16907073/test1a.py.gz)

tcaduser, 2024-09-06:
Hello @Alex236842,

I am able to get much better convergence, by adding these lines before the initial drift-diffusion solution:
```
###
### Drift diffusion simulation at equilibrium
###

diode_common.DriftDiffusionInitialSolution(device, region)
opts = get_equation_command(name="PotentialEquation", device=device, region=region)
opts['variable_update']="default"
equation(**opts)
```
Adding the "variable_update" option changes the behavior so that voltage limiting is turned off.

Alex236842, 2024-09-06:
Dear Juan,thank you for a very fast response which may be solves this test. I will try it out.Regards, Alex--Gesendet mit der GMX Mail App

Alex236842, 2024-09-09:
> Hi @Alex236842,
>
> Thanks for your report. I am running version 2.8.2 on my macbook and I am seeing:

Dear Juan,

 

first of all, I find it very good that your Mac version gives an excellent convergence and accuracy.

I included the changes that you recommended in my Linux devsim version 2.7.0, but it did not

bring the expected result.

Therefore, I upraded devsim to 2.8.2.

The new devsim version did not bring the convergence improvement either.

So, it seems the problem is related to Linux.

 

There is one place in the output after which the error in the equation for electrons becomes very large,

of about 1e47 and after this nothing helps.

 

Here is this fragment from vevsim output:

# output from solve(type="dc", absolute_error=1e-2, relative_error=1e-2, maximum_iterations=200)

# looks nice, a convergence is achieved for PotentialEquation:

Iteration: 89
  Device: "diode3d"    RelError: 9.91956e-03    AbsError: 1.04878e-03
    Region: "Bulk"    RelError: 9.91956e-03    AbsError: 1.04878e-03
      Equation: "PotentialEquation"    RelError: 9.91956e-03    AbsError: 1.04878e-03

 

# But then these warnings follow:
Warning: Replacing equation with equation of the same name.
Region: Bulk, Equation: PotentialEquation, Variable: Potential
Warning: Replacing equation with equation of the same name.
Region: Bulk, Equation: PotentialEquation, Variable: Potential
number of equations 24

 

# And the following statement

# solve(type="dc", absolute_error=1e12, relative_error=1, maximum_iterations=100)
# shows an AbsError: 4.76837e+47
Iteration: 0
  Device: "diode3d"    RelError: 1.00001e+03    AbsError: 4.76837e+47
    Region: "Bulk"    RelError: 1.00001e+03    AbsError: 4.76837e+47
      Equation: "ElectronContinuityEquation"    RelError: 1.00000e+00    AbsError: 4.76837e+47
      Equation: "HoleContinuityEquation"    RelError: 9.99000e+02    AbsError: 1.68200e+08
      Equation: "PotentialEquation"    RelError: 1.46213e-02    AbsError: 1.56882e-03
Iteration: 1
  Device: "diode3d"    RelError: 1.99801e+03    AbsError: 4.76360e+47
    Region: "Bulk"    RelError: 1.99801e+03    AbsError: 4.76360e+47
      Equation: "ElectronContinuityEquation"    RelError: 9.99000e+02    AbsError: 4.76360e+47
      Equation: "HoleContinuityEquation"    RelError: 9.99000e+02    AbsError: 1.68200e+05
      Equation: "PotentialEquation"    RelError: 1.43574e-02    AbsError: 1.56295e-03

 

Could you check what happens here in the Linux version 2.8.2 python 3.7?

 

I think it will also be useful for the Forum, especially when we find a Linux solution.

 

Thank you in advance,

 

Alex

 
 

Gesendet: Freitag, 06. September 2024 um 17:14 Uhr

bot       +0.000e+00	-6.325e-01	-1.619e-23	-6.325e-01
top       +1.000e+00	+6.325e-01	+1.619e-23	+6.325e-01

which is not accurate. However, if I add these lines after the loop:

solve(type="dc", absolute_error=5e12, relative_error=1e-10, maximum_iterations=100)
print_all_currents()

I then get this result after 8 iterations:

bot       +0.000e+00	-1.000e+00	-2.560e-23	-1.000e+00
top       +1.000e+00	+1.000e+00	+2.560e-23	+1.000e+00

which is a much better result.

If you haven't already, please feel free to participate in our online forum:
https://forum.devsim.org

To upgrade devsim:

pip install devsim --upgrade

And please update this issue if you do not see an improved result. I may then need to see if there are any issues that are Linux specific.

tcaduser, 2024-09-09:
The test case runs on an aarch64 docker instance with the default UMFPACK solver.

tcaduser, 2024-09-09:
Hi @Alex236842 

On `x86_64`, I am able to reproduce your issue.

The following works for me:
```
python -mdevsim.umfpack.umfshim test1a.py
```
so that `umfpack` is selected over the Intel MKL Pardiso.

Alternatively, the Intel MKL can be removed, and an alternative numeric library can be installed:
```
conda remove mkl
conda install openblas
python test1a.py
```

but that has negative consequences with respect to the usage of `numpy` and `matplotlib` and the versions that Anaconda might make available.  Using the `umfshim` will use the BLAS/LAPACK routines from Intel, but resort to UMFPACK for solving.

I am not sure if I will be able to fix the call to the Intel MKL.

Please try my suggested workaround and let me know what you think would be the best solution.

Also, if you are aware of any solvers that `devsim` should be using in addition to `umfpack` or `pardiso`, I would be grateful.  `SuperLU` is deprecated and that solver is not available using the binary releases.

Alex236842, 2024-09-11:
I confirm that the call
python -mdevsim.umfpack.umfshim test1a.py
using DEVSIM 2.8.2 on Linux produces a fast convergence and reproduces 1A result with an ideal accuracy.
Solution:
Contact   Voltage        Electron    Hole        Total     
                         Current     Current     Current
bot       +0.000e+00	-1.000e+00	-2.560e-23	-1.000e+00
top       +1.000e+00	+1.000e+00	+2.560e-23	+1.000e+00

closed 2024-09-11 by tcaduser