devsim/devsim #152: memory leak

heheda2001123, 2024-10-15:
An memory leak while using Gmsh meshed device to iteratively solve a device biased at 150V with 0.1V step, 200M memory leak per step.

[CMOS_strip.zip](https://github.com/user-attachments/files/17373142/CMOS_strip.zip)

heheda2001123, 2024-10-15:
this memory leak is found in version 2.6.3, though it continues to exist in the latest version.

in 2.6.3 the program runs until the memory exceeds 16G, but in this version it meets an convergence failure, though before the failure the memory used increases from 2G to 8G.

tcaduser, 2024-10-15:
Also, could you please give me the output of:
```
import devsim
print(devsim.get_parameter(name='info'))
```

heheda2001123, 2024-10-15:
On debian 11.7; 

Python 3.9.2 (default, Feb 28 2021, 17:03:44) 
[GCC 10.2.1 20210110] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import devsim
>>> print(devsim.get_parameter(name='info'))
{'copyright': 'Copyright © 2009-2024 DEVSIM LLC', 'direct_solver': ('mkl_pardiso',), 'extended_precision': True, 'license': 'Apache License, Version 2.0', 'math_libraries': ('libmkl_rt.so.2',), 'mkl_version': 'Intel(R) oneAPI Math Kernel Library Version 2024.2.2-Product Build 20240823 for Intel(R) 64 architecture applications', 'version': '2.8.4', 'website': 'https://devsim.org'}
>>>

tcaduser, 2024-10-15:
Hello @heheda2001123 

I am currently running your script using:
```
valgrind --tool=memcheck --leak-check=full python solver_section.py  |& tee file2.out
```

where valgrind is a memory leak detector.  An initial run showed that there were blocks of memory still being referenced when I terminated the program early.

```
==5065== LEAK SUMMARY:
==5065==    definitely lost: 0 bytes in 0 blocks
==5065==    indirectly lost: 0 bytes in 0 blocks
==5065==      possibly lost: 24,785,723 bytes in 152 blocks
==5065==    still reachable: 385,311,627 bytes in 1,482,685 blocks
==5065==                       of which reachable via heuristic:
==5065==                         stdstring          : 150 bytes in 2 blocks
==5065==         suppressed: 0 bytes in 0 blocks
==5065== Reachable blocks (those to which a pointer was found) are not shown.
==5065== To see them, rerun with: --leak-check=full --show-leak-kinds=all
==5065== 
==5065== Use --track-origins=yes to see where uninitialised values come from
==5065== For lists of detected and suppressed errors, rerun with: -s
==5065== ERROR SUMMARY: 127 errors from 41 contexts (suppressed: 0 from 0)
```

Running with valgrind is much slower.

I am now rerunning the script up to `0.1` V, and with the:
```
devsim.reset_devsim()
```
command at the very end of the script in the hopes of getting a full report on whether blocks of memory are actually being lost.

The "possibly lost" category is reported mostly MKL (24 MB).  The largest block is:

```
==5065== 9,589,824 bytes in 2 blocks are possibly lost in loss record 5,171 of 5,177
==5065==    at 0x4846828: malloc (in /usr/libexec/valgrind/vgpreload_memcheck-amd64-linux.so)
==5065==    by 0x6DDE972: mm_internal_malloc (in /root/venv/lib/libmkl_core.so.2)
==5065==    by 0x6DDCF4F: mkl_serv_allocate (in /root/venv/lib/libmkl_core.so.2)
==5065==    by 0x6BAF9F54: mkl_blas_avx2_dgemm_get_bufs (in /root/venv/lib/libmkl_avx2.so.2)
==5065==    by 0x6BAFB6A5: mkl_blas_avx2_d_generic_fullacopybcopy (in /root/venv/lib/libmkl_avx2.so.2)
==5065==    by 0x6BB0CF54: mkl_blas_avx2_xdgemm (in /root/venv/lib/libmkl_avx2.so.2)
==5065==    by 0x83D8B3C: mkl_pds_lp64_upd_real_block (in /root/venv/lib/libmkl_core.so.2)
==5065==    by 0xC4B9C25: mkl_pds_lp64_blkl_ll_real.extracted (in /root/venv/lib/libmkl_intel_thread.so.2)
==5065==    by 0xAD36C42: __kmp_invoke_microtask (in /root/venv/lib/libiomp5.so)
==5065==    by 0xACBEFD0: __kmp_serial_fork_call(ident*, int, fork_context_e, int, void (*)(int*, int*, ...), int (*)(int), kmp_info*, kmp_team*, ompt_data_t*, void**, ompt_data_t**, __va_list_tag (*) [1]) (kmp_runtime.cpp:2089)
==5065==    by 0xACBD715: __kmp_fork_call (kmp_runtime.cpp:2405)
==5065==    by 0xAC7FC87: __kmpc_fork_call (kmp_csupport.cpp:392)
```

The "still reachable" category is about (385 MB) and finding its source requires the new run with ``--show-leak-kinds=all``.

I am currently running this in an Ubuntu 24.04 docker image with Python 3.12.

I will let you know how things are proceeding.  Please try `valgrind` if you get a chance, if it is available on your system.

tcaduser, 2024-10-15:
```
pip install devsim mkl matplotlib numpy psutil
```

tcaduser, 2024-10-18:
[debian2.txt](https://github.com/user-attachments/files/17436502/debian2.txt)
[ubuntu2.txt](https://github.com/user-attachments/files/17436503/ubuntu2.txt)

Running to 2 V on a Debian 11.11 system running Python 3.9 and an Ubuntu 24.04 system running Python 3.12.  I am not really seeing a large memory leak in valgrind.  Please let me know your thoughts.

heheda2001123, 2024-10-26:
Sorry for late response, I have tried using valgrind to get the output of a full run, but since the thread was auto killed I can't get corresponding valgrind report. i will try running to 4 V (before it will be killed) and show the difference today.

By the way the package ``` psutil ``` is imported for memory monitoring too, process.memory_info().rss and process.memory_info().vms will increase from 8 G at 2 V to 16 G at 4 V.

I wonder if "normally allocated" memory space would count as memory leak by valgrind; if no, it seems to be rather a "memory allocating strategy" problem for devsim.

tcaduser, 2024-10-26:
Hi @heheda2001123,

Thanks for the data point.  This is just a guess, but I suspect your system may have the Linux Out-Of-Memory Killer settings enabled.  For example:
https://www.oracle.com/technical-resources/articles/it-infrastructure/dev-oom-killer.html

While your system may have enough memory, the administrative settings may be restricting its use.

This may also be coupled with the use of cgroups to limit per user authorization to memory access. 

I will have access to a Linux box early next week, and I will report if I am able to reproduce your issue between 2 and 4 V running without valgrind.

Since you are many large bias steps, have you tried changing the variable update strategy I suggest here:
https://forum.devsim.org/t/memory-leak-after-devsim-solve/348

heheda2001123, 2024-10-27:
I have changed the solving iteration strategy but did not get any considerable difference. When I tried to record valgrind log, the process was killed at 0.1V, way before I expected.

tcaduser, 2024-10-27:
valgrind creates a virtual machine that takes up memory.  I was able to run to 0.2 V.  If we are running the same example, it suggests there are administrator imposed resource limits on it.

heheda2001123, 2024-10-29:
The server I use allows me to allocate an arbitrary large amount of memory (if capable). The voltage when the job is killed grows linearly with the memory amount (I have tried from 8G to 30G), consistent with what ```psutil.Process().memory_info().rss``` said.

I also tried to run the example on UTM (debian 12.7, 8G memory) in my machine, and still, the thread is killed when ```psutil.Process().memory_info().rss``` reaches 8G. That is, I am sure the thread is killed because it reached a natural memory limit, not an artificial one. 

Since the goal voltage of my device is 150V, to linearly extrapolate, I cannot expect any kind of usual machine could afford me a memory of half a terabyte...

tcaduser, 2024-10-29:
I suspect it is the overcommit memory settings on your machine.  Can you share this output:
```
cat /proc/sys/vm/overcommit_memory 
```

tcaduser, 2024-10-29:
I am running you example now to 150 V and we'll see how far it goes on a 32GB machine

tcaduser, 2024-10-29:
I ended up killing the example after 2.7 V because it was not converging after 807 iterations.

I do not recommend more than about 30 iterations before reducing the voltage step, or changing the convergence criteria.  You may also want to see if your mesh needs more refinement.

The last memory usage statement:
```
Memory usage before calling loop_solver: RSS = 6450.14 MB, VMS = 7515.97 MB
```

tcaduser, 2024-10-29:
with a relative error of 1e-7, the simulation is not converging after 2.9 V

tcaduser, 2024-10-29:
I am having more success with:
```
python -mdevsim.umfpack.umfshim solver_section.py
```

It is still running after 4.6 V.
```
Memory usage after calling loop_solver: RSS = 1216.31 MB, VMS = 2265.61 MB
Memory usage before calling loop_solver: RSS = 1216.31 MB, VMS = 2265.61 MB
```

I will let you know how far it gets.

tcaduser, 2024-10-29:
Using the `umfshim` to load `umfpack` instead of `mkl_pardiso`, it gets to 10.2 V
```
top	10.200000000000001	1.1098329854086965e-11	8.092493536109776e-16	1.1099139103440575e-11
```
and then fails at 10.3 V for 999 iterations.

```
Memory usage after calling loop_solver: RSS = 1214.32 MB, VMS = 2263.54 MB
Memory usage before calling loop_solver: RSS = 1214.32 MB, VMS = 2263.54 MB
```

heheda2001123, 2024-10-30:
> I suspect it is the overcommit memory settings on your machine. Can you share this output:
> 
> ```
> cat /proc/sys/vm/overcommit_memory 
> ```

it returns 0.

heheda2001123, 2024-10-30:
> Using the `umfshim` to load `umfpack` instead of `mkl_pardiso`, it gets to 10.2 V
>
> ```
> top	10.200000000000001	1.1098329854086965e-11	8.092493536109776e-16	1.1099139103440575e-11
> ```
>
> and then fails [...]

!!!

So we could expect with `umfshim` the memory problem could be eliminated, and the problem is reduced to a mesh-tuning one. Many thanks to you during the long time.

tcaduser, 2024-10-30:
> > I suspect it is the overcommit memory settings on your machine. Can you share this output:
> > ```
> > cat /proc/sys/vm/overcommit_memory 
> > ```
> 
> it returns 0.

It looks like the overcommit_memory is not the issue.  It may just be the limited resources on having 8 GB memory for the size of the simulation you are trying to solve with the Intel MKL solver

tcaduser, 2024-10-30:
> !!!
>
> So we could expect with `umfshim` the memory problem could be eliminated, and the problem is reduced to a mesh-tuning one. Many thanks to you during the long time.

Please try it out and see if it helps for your situation.  One thing you can try is to save the mesh at the last bias step that is successful.  Perhaps something like:

```
try:
    devsim.solve( )
except devsim.error as msg:
    devsim.write_devices(file="test.tec", type="tecplot")
    raise
```

You can then look at the visualization file, and check the quality of the results.

tcaduser, 2024-10-30:
It is possible to create a background mesh to aid Gmsh mesh refinement based on simulation results.  Please see:
https://github.com/devsim/devsim_misc/tree/main/refinement

for an example.

yh-kwok, 2024-12-13:
Hi all, I also notice that the memory usage keeps increasing every time the "solve" command is called.
I am able to reproduce it using the devsim diode_2d example, by increasing the mesh density and keep calling solve.
(DEVSIM 2.9.1 installed using pip, on ubuntu 22.04)

For 100 solves, memory usage remains stable using UMFPACK, but increases indefinitely using MKL solver.
I tried setting the environment variable MKL_DISABLE_FAST_MM to 1 but it doesn't seem to help.

Attached are the modified scripts and memory logs. Hope it helps.
[diode_2d_memory.zip](https://github.com/user-attachments/files/18124114/diode_2d_memory.zip)

tcaduser, 2024-12-14:
Hi @yh-kwok,

Can you see if updating the MKL version helps:
```
pip install mkl --upgrade
```
since I see you are using the 2023.2 version.  If you are using Anaconda python, please try the appropriate update command.

Please let me know which version of Python (Anaconda, Ubuntu, . . .) you are using and I'll test it this weekend.

yh-kwok, 2024-12-16:
I upgraded MKL to 2024.2.2 but the issue persists.
I am using Python 3.10.13, mamba 1.5.1, conda 23.3.1.

tcaduser, 2024-12-17:
Hi @yh-kwok,

Thanks for the detailed information.  I will be able to take a look at it this weekend.  Until then, could you please report any environment variables you may have set, such as the output from:
```
env | grep MALLOC_
env | grep M_
```
which has some control over memory location.

tcaduser, 2024-12-17:
also, if I need to use mamba to reproduce your environment, please provide instructions.

tcaduser, 2024-12-17:
as suggested:
https://github.com/pytorch/pytorch/issues/64535

Please try also setting:
```
export MKL_NUM_THREADS=1
```
and see if that helps with the issue

yh-kwok, 2024-12-20:
mamba is just a faster reimplementation of conda package manager. It shouldn't matter.

Both commands give empty result:
```
env | grep MALLOC_
env | grep M_
```

tcaduser, 2024-12-23:
> as suggested: [pytorch/pytorch#64535](https://github.com/pytorch/pytorch/issues/64535)
> 
> Please try also setting:
> 
> ```
> export MKL_NUM_THREADS=1
> ```
> 
> and see if that helps with the issue

Hi @yh-kwok, did you get a chance to try this?

yh-kwok, 2025-01-02:
In my case, leak still happens with
```
export MKL_NUM_THREADS=1
export OMP_NUM_THREADS=1
```